MarlinFirmware/Marlin #16903: [BUG] Z probe offset doesn't work live

Liger0, 2020-02-19:
Bug Description

The zprobe offset doesn't work in live

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4226613/Marlin.zip)
Today's bugfix.

### Steps to Reproduce

1. Auto home
2. Go to Config-> Probe z offset
3. Change the value

**Expected behavior:** Move the z axis along to respect the offset

**Actual behavior:** Nothing happens until I rehome

#### Additional Information

With old version of bugfix from january it worked using same settings.

thisiskeithb, 2020-02-19:
I reported this as well: https://github.com/MarlinFirmware/Marlin/issues/16889

Liger0, 2020-02-19:
I see, I didn't notice it. I'm closing this.

closed 2020-02-19 by Liger0