MarlinFirmware/Marlin #16904: Bugg when removing and inserting sd card, won't print properly(Marlin 2.0.3)

alfonsg00, 2020-02-19:
I have a issue with Marlin 2.0.3. I'm using an Ender 3 with the cheetah 1.1b board. If i remove and insert the sd card before starting a print the printhead acts strange and won't print properly. You can hear a "clonk" after re-inserting the sd card from the z axis. If you initiate eeprom before printing it will print normally. Tried to use EEPROM_AUTO_INIT but didn't make any difference.

alfonsg00, 2020-02-19:
Link for videos of the issue: https://youtu.be/didVrh2fIPc
https://youtu.be/KQLpTWLgGNE

andysport, 2020-02-20:
I have the same printer, board and marlin version too. Yes, I can confirm that bug. When I insert the SD, I can hear a noise and the stepper drives move with high speed and noise. If I restore failsafe, the issues disappears.

Wotovski, 2020-02-21:
I have the same problem. I can't start any Printjob from my SD Cards. My Printer is a Anycubic Kossel.

Hagbard-84, 2020-02-23:
I have the same problem. Ender 3, cheetah 1.1, Marlin 2.0.3. befor this happened the printer reported "no media" "media inserted".

trouch, 2020-03-29:
Issue still is happening with current bugfix-2.0.x branch and doesn't seem to be related to the previous SD detect bug. 
I also have Fysetc Cheetah 1.1 board (STM32F1 + TMC2209/UART) with Flash EEPROM enabled by the way.

It is easy to reproduce by inserting the SD _after_ powering on.
Steppers/drivers configuration get somehow corrupted, X axis becomes super fast and loud, Y super slow and Homing fails then.

There is no issue when inserting the card _before_ powering on the printer.
Workaround is power-cycling the printer after SD card insertion, but it is not convenient.

I tried to debug but so far unlucky.

trouch, 2020-03-30:
More testing ...

I've set `#define SDCARD_CONNECTION ONBOARD` in order to disable card detect and auto-mount, and issue is still is occurring.
As @alfonsg00 said, we can clearly a _clonk_ noise - motors being disabled - at the moment we insert the card, regardless of `SDCARD_CONNECTION` and auto-mount.

At this point I highly suspect an hardware issue or STM32(F1) specific one which break something when SD lines are hotplugged.

thinkyhead, 2020-03-31:
> I really think about an hardware design issue since only Cheetah users seem affected.

That is a distinct possibility. Any buzz about this in Cheetah user groups?

thinkyhead, 2020-03-31:
Please continue discussion over at #17333.

closed 2020-03-31 by thinkyhead