MarlinFirmware/Marlin #16905: [FR] TMC2209 Support physical endstop switches without removing diag pin

stantond, 2020-02-19:
### Description

1. I want to use my physical endstops with 2209 drivers, so that I can home with consistent reliability. There is no way sensorless homing is ready for the masses.

2. I want to to the above without cutting the diag pins, so that I can still use sensorless homing when the feature is more reliable.

### Feature Workflow

1. Sensorless homing is disabled
2. 2209 drivers are installed and no pins are removed
3. Endstops are installed
4. Homing uses the physical endstops

#### Additional Information

* I've attached my current configuration. Currently, endstops are ignored. I'm happy try try tweaks and changes if we think there's a chance we can get this working.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4227480/Marlin.zip)

ManuelMcLure, 2020-02-19:
On the SKR 1.4 it is physically impossible to use the endstop connectors with the diag pins intact due to the electrical design. The diag pins are hardwired to the end stop inputs so they will always override the switch values. There is nothing that can be done on the Marlin side to make this work with the defined end stop pins. The SKR 1.3 has jumpers that allow disconnecting the diag pins, but the SKR 1.4 does not.

The only possible workaround without cutting the diag pins is to wire the endstop switches to different pins (for example on the AUX connectors). This will require setting up a pull up and possibly a filter capacitor for each endstop.

thisiskeithb, 2020-02-19:
You'll need to reach out to BigTreeTech and suggest they add diag jumpers on the next revision since the diag lines from each driver are connected to the endstops on SKR 1.4 (and most of the newer boards from them).

This is why BigTreeTech tells you to snip the pin or desolder it and Marlin can't override it.

stantond, 2020-02-19:
I understand that the diag pins are connected to the corresponding end stop pins, but not why this matters. What exactly are the drivers doing that makes the physical endstop undetectable?

Grogyan, 2020-02-19:
There seems to be a misunderstanding,
The newer versions of the board are specifically tailored for sensorless homing, without physical X and Y Endstops, but when using non stallguard drivers, physical endstops have to be used.

Is there any reason as to why the SKR1.4 turbo must be used over the the existing, and quite capable, SKR1.3?

stantond, 2020-02-19:
Yes, I own a SKR 1.4 Turbo, and believed that sensorless homing would work reliably and consistently enough for me to use it instead of endstops. This is replacing a Creality 1.1.5 board, not an SKR 1.3

ManuelMcLure, 2020-02-19:
The end stop connector has a pull-up resistor and a cap. That means that if nothing is connected it will set the end stop input pin to HIGH (+VCC). Usually an end stop switch will, when triggered, connect the pin to ground for a LOW signal.

The DIAG pin on the TMC2209 outputs LOW when a stall has not been detected. This means that the end stop pin is effectively tied to ground just like when the end stop switch is triggered.

Even if you try to connect up the end stop switch to pull the pin to +VCC when triggered, you will be fighting with the TMC2209 which is trying to pull the pin low. At best it will be very unreliable, at worst you could damage the TMC2209.

stantond, 2020-02-19:
That makes sense, thanks. Maybe I'll record the inconsistent homing behavior I'm seeing with sensorless homing and open an issue for that before cutting bits of my drivers off.

closed 2020-02-19 by stantond