MarlinFirmware/Marlin #16907: [BUG] (short description)Pausing or Cancelling prints from Octoprint leaves printer in "Printing..."state with Marlin 2.0.3

Thinkersbluff, 2020-02-20:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-02-20:
Suspect this is issue  #16749, this has been fixed in bugfix and will be in next release.

GMagician, 2020-02-20:
and by #16849, see issue #16893

Thinkersbluff, 2020-02-21:
Thanks for the cross-references, @ellensp and @GMagician.  Evidently, my strategies for checking whether an issue is already identified still need work.  I spent a lot of time trying to deduce whether my issue was an Octoprint issue, a BTT SKR Mini issue, or a serial interface issue.  By the time I thought of comparing 2.0.1 to 2.0.3, I was focused specifically on Octoprint, not the more generic term "host".
Yes, using the latest bug-fix version today, when the Host Cancel print command includes turning-off the heaters, the action of Cancelling from the Host interface does now cause the printer to stop the job.
I do find, however, that Pausing a job from the Host simply pauses the datastream and the printer just stops moving when it has drained the buffer.  The printer does not realize that the job has been paused. 
Maybe that is an issue for the Host developers to address...  At least Octoprint allows me to specify gcodes to send when Canceling a job.  I will go ahead and close this issue, assuming that the current behaviour is a feature, not a bug, from Marlin's point of view.

closed 2020-02-21 by Thinkersbluff