MarlinFirmware/Marlin #16908: [BUG] M0 no longer works since commit 0ce83dee9560fc30f27fdddc450d3861725d9a22

ellensp, 2020-02-20:
### Bug Description

M0 no longer puts the machine into a waiting state since commit 0ce83dee9560fc30f27fdddc450d3861725d9a22

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4229117/config.zip)

### Steps to Reproduce

1. Enable EMERGENCY_PARSER
2. Enable REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
3. Set BOARD_BTT_SKR_V1_3  (but effects all motherboards) 
4. set  SERIAL_PORT -1 just to make SKR  happy.
5. start up printer
6. Connect to printer
7. Send a "M0 hello" to printer

**Expected behavior:** 

LCD status should change to hello and the machine should be paused, sending  "echo:busy: paused for user" to the console

**Actual behavior:** 

Nothing. The status is not updated and pause state is not set.

#### Additional Information

Found while looking into #16902 

### The issue

Logic in Marlin/src/gcode/lcd/M0_M1.cpp line 105: Does not work as it did before the commit.
Original code is: `while (wait_for_user && (ms > 0 || PENDING(millis(), ms))) idle();`
Changing it to `while (wait_for_user || (ms > 0 && PENDING(millis(), ms))) idle();` seem make it work as originally intended

TheNikomo, 2020-02-20:
Running `62b9d7dc73e34437ebfe52ad7324b37698569275` on my SKR 1.3 with a different hardware configuration, I can confirm that M0 does not halt the printer.

Sending `M0 Hello` to printer causes nothing to happen on the display, and machine can be operated as normal, even through serial - sending `G28` after `M0` without touching the printer causes printer to home.

Attaching configuration files in case those shed any light.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4230705/Configurations.zip)

Moyster, 2020-02-22:
I guess this missed the 2.0.4.1 release :/ @ellensp maybe you should push a PR with your while loop revert ? GJ finding this regression :)

ellensp, 2020-02-22:
My fork was in disarray preventing useful PR's, but sorted now and PR has been created

thinkyhead, 2020-02-23:
Now included in 2.0.4.2.

closed 2020-02-23 by thinkyhead