MarlinFirmware/Marlin #16910: [BUG] UBL grid vertices visible in prints

halninekilo, 2020-02-20:
### Bug Description

UBL grid vertices visible in prints.

I saw https://github.com/MarlinFirmware/Marlin/issues/10171 but don't think it applies here since I set jerk to `M205 X10 Y10 Z0.3 E5.0;` and I am using JD anyway (default values) - why would the hotend pause during a straight move?

### My Configurations

[configuration_headerfiles.zip](https://github.com/MarlinFirmware/Marlin/files/4231408/configuration_headerfiles.zip)

### Steps to Reproduce

1. Enable UBL (`G29 P1; G29 F 10.0; G29 A;`)
2. Start a print and see blobs emerging at the mesh vertices.
3. Change `GRID_MAX_POINTS_X` and repeat steps 1-2 to verify correlation between grid-points and artefacts in print.

#### Additional Information
Marlin Bugfix (020004), Ender 2, SKR 1.3, TMC2209

[image]
(`GRID_MAX_POINTS_X 10`)

[image]
(`GRID_MAX_POINTS_X 4`)

closed 2020-02-20 by halninekilo

rado79, 2020-02-20:
confirmed. there are spots in infill, in my case i can observe on extrudergear that linear advance does release pressure on some spots, every layer on same place. really hard visible on surface.

halninekilo, 2020-02-20:
Usererror in my case (invalid mesh) - filling in two outlines of missing points via `G29 P3; G29 P3;` seems to have solved the issue.

rado79, 2020-02-21:
There are no missing probe points in my mesh.