MarlinFirmware/Marlin #16912: [FR] (feature request title)

gordio, 2020-02-20:
Is it possible to disable (hide) this menu item `#define MSG_CHANGE_MEDIA                    _UxGT("Change Media")`

### Description

I have only SD Card and that item just decrease usability of menu.

### Feature Workflow

1. `#define HIDE_CHANGE_MEDIA_ITEM`
2. Compile and Item is gone from LCD menu

Anthrix, 2020-02-29:
Not working fo me on bugfix-2.0.x still there after defining HIDE_CHANGE_MEDIA_ITEM

ellensp, 2020-03-01:
This is a FR, a feature request. It is not implemented yet.

Power3DPrinting, 2020-04-01:
Bump, I want the change media option gone as well, only using an SD card so change media does nothing

qwewer0, 2021-01-01:
https://github.com/MarlinFirmware/Marlin/issues/20309 Related?

andiweli, 2022-09-18:
Is this already solved?
I find this menu item also disturbing, any way to hide it?

wuspy, 2022-10-11:
You can hide it pretty easily by just removing/commenting this line of code in `menu_main.cpp` (currently line 260 on 2.1.x)

Why there isn't a config option for this I'm not sure. I've never needed to use it in the 4 years I've been running marlin on my printer.

```diff
    auto sdcard_menu_items = [&]{
      #if ENABLED(MENU_ADDAUTOSTART)
        ACTION_ITEM(MSG_RUN_AUTO_FILES, card.autofile_begin); // Run Auto Files
      #endif

      if (card_detected) {
        if (!card_open) {
          #if HAS_SD_DETECT
-            GCODES_ITEM(MSG_CHANGE_MEDIA, F("M21"));        // M21 Change Media
          #else                                             // - or -
            ACTION_ITEM(MSG_RELEASE_MEDIA, []{              // M22 Release Media
              queue.inject(F("M22"));
              #if ENABLED(TFT_COLOR_UI)
                // Menu display issue on item removal with multi language selection menu
                if (encoderTopLine > 0) encoderTopLine--;
                ui.refresh(LCDVIEW_CALL_REDRAW_NEXT);
              #endif
            });
          #endif
          SUBMENU(MSG_MEDIA_MENU, MEDIA_MENU_GATEWAY);      // Media Menu (or Password First)
        }
      }
      else {
        #if HAS_SD_DETECT
          ACTION_ITEM(MSG_NO_MEDIA, nullptr);               // "No Media"
        #else
          GCODES_ITEM(MSG_ATTACH_MEDIA, F("M21"));          // M21 Attach Media
        #endif
      }
    };
```