MarlinFirmware/Marlin #16915: [BUG] Macro E_AXIS_HAS(T) does not resolve E0_HARDWARE_SERIAL properly for ANY_AXIS_HAS(SW_SERIAL)

vivian-ng, 2020-02-21:
### Bug Description

`drivers.h` fails to recognize `#define E0_HARDWARE_SERIAL Serial1` declared in `Configuration_adv.h`, resulting in fail of sanity check on ESP32 platform. Probably because the macro `E_AXIS_HAS(T)` is not resolving properly.

### My Configurations

In `Configuration.h`:
```
#define MOTHERBOARD BOARD_MRR_ESPA

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988
```

In `Configuration_adv.h`:
```
  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 1
  #define  Z_SLAVE_ADDRESS 2
  #define X2_SLAVE_ADDRESS 0
  #define Y2_SLAVE_ADDRESS 0
  #define Z2_SLAVE_ADDRESS 0
  #define Z3_SLAVE_ADDRESS 0
  #define Z4_SLAVE_ADDRESS 0
  #define E0_SLAVE_ADDRESS 3
  #define E1_SLAVE_ADDRESS 0
  #define E2_SLAVE_ADDRESS 0
  #define E3_SLAVE_ADDRESS 0
  #define E4_SLAVE_ADDRESS 0
  #define E5_SLAVE_ADDRESS 0
  #define E6_SLAVE_ADDRESS 0
  #define E7_SLAVE_ADDRESS 0

  #define X_HARDWARE_SERIAL Serial1
  #define Y_HARDWARE_SERIAL Serial1
  #define Z_HARDWARE_SERIAL Serial1
  #define E0_HARDWARE_SERIAL Serial1
  #define TMC_BAUD_RATE 115200, SERIAL_8N1, 21, 22
```

### Steps to Reproduce

1. Make above declarations in Configuration.h and Configuration_adv.h
2. Build

**Expected behavior:** Build complete without failure.

**Actual behavior:** Build fails at 
```
Compiling .pio/build/esp32/src/src/HAL/HAL_ESP32/i2s.cpp.o
In file included from Marlin/src/HAL/HAL_ESP32/../../inc/MarlinConfig.h:39:0,
                 from Marlin/src/HAL/HAL_ESP32/timers.h:29,
                 from Marlin/src/HAL/HAL_ESP32/HAL.h:37,
                 from Marlin/src/HAL/HAL_ESP32/HAL.cpp:25:
Marlin/src/HAL/HAL_ESP32/../../inc/../HAL/HAL_ESP32/inc/SanityCheck.h:33:4: error: #error "TMC220x Software Serial is not supported on this platform."
   #error "TMC220x Software Serial is not supported on this platform."
    ^
In file included from Marlin/src/HAL/HAL_ESP32/../../inc/MarlinConfig.h:39:0,
                 from Marlin/src/HAL/HAL_ESP32/timers.h:29,
                 from Marlin/src/HAL/HAL_ESP32/HAL.h:37,
                 from Marlin/src/HAL/HAL_ESP32/HAL_SPI.cpp:26:
Marlin/src/HAL/HAL_ESP32/../../inc/../HAL/HAL_ESP32/inc/SanityCheck.h:33:4: error: #error "TMC220x Software Serial is not supported on this platform."
   #error "TMC220x Software Serial is not supported on this platform."
    ^
In file included from Marlin/src/HAL/HAL_ESP32/../../inc/MarlinConfig.h:39:0,
                 from Marlin/src/HAL/HAL_ESP32/Servo.cpp:24:
Marlin/src/HAL/HAL_ESP32/../../inc/../HAL/HAL_ESP32/inc/SanityCheck.h:33:4: error: #error "TMC220x Software Serial is not supported on this platform."
   #error "TMC220x Software Serial is not supported on this platform."
    ^
*** [.pio/build/esp32/src/src/HAL/HAL_ESP32/HAL.cpp.o] Error 1
*** [.pio/build/esp32/src/src/HAL/HAL_ESP32/HAL_SPI.cpp.o] Error 1
*** [.pio/build/esp32/src/src/HAL/HAL_ESP32/Servo.cpp.o] Error 1
In file included from Marlin/src/HAL/HAL_ESP32/../../module/../inc/MarlinConfig.h:39:0,
                 from Marlin/src/HAL/HAL_ESP32/../../module/stepper.h:44,
                 from Marlin/src/HAL/HAL_ESP32/i2s.cpp:33:
Marlin/src/HAL/HAL_ESP32/../../module/../inc/../HAL/HAL_ESP32/inc/SanityCheck.h:33:4: error: #error "TMC220x Software Serial is not supported on this platform."
   #error "TMC220x Software Serial is not supported on this platform."
    ^
*** [.pio/build/esp32/src/src/HAL/HAL_ESP32/i2s.cpp.o] Error 1
```
#### Additional Information

In `drivers.h`:
Check for TMC software serial is done via
```
#define HAS_TMC_SW_SERIAL  ANY_AXIS_HAS(SW_SERIAL)
```
However, 
```
#define E_AXIS_HAS(T)   (0 RREPEAT2(E_STEPPERS, _OR_EAH, T))

#define ANY_AXIS_HAS(T) (    AXIS_HAS_##T(X)  || AXIS_HAS_##T(X2) \
                          || AXIS_HAS_##T(Y)  || AXIS_HAS_##T(Y2) \
                          || AXIS_HAS_##T(Z)  || AXIS_HAS_##T(Z2) \
                          || AXIS_HAS_##T(Z3) || AXIS_HAS_##T(Z4) \
                          || AXIS_HAS_##T(Z3) || AXIS_HAS_##T(Z4) \
                          || E_AXIS_HAS(T) )
```
the macro `E_AXIS_HAS(T)` fails to properly recognize that E0_HARDWARE_SERIAL has been defined.

Changing `E_AXIS_HAS(T)` to:
```
#define E_AXIS_HAS(T)   (0 RREPEAT2(E_STEPPERS, _OR_EAH, T))

#define ANY_AXIS_HAS(T) (    AXIS_HAS_##T(X)  || AXIS_HAS_##T(X2) \
                          || AXIS_HAS_##T(Y)  || AXIS_HAS_##T(Y2) \
                          || AXIS_HAS_##T(Z)  || AXIS_HAS_##T(Z2) \
                          || AXIS_HAS_##T(Z3) || AXIS_HAS_##T(Z4) \
                          || AXIS_HAS_##T(Z3) || AXIS_HAS_##T(Z4) \
                          || AXIS_HAS_##T(E0) || AXIS_HAS_##T(E1) \
                          || AXIS_HAS_##T(E2) || AXIS_HAS_##T(E3) \
                          || AXIS_HAS_##T(E4) || AXIS_HAS_##T(E5) \
                          || AXIS_HAS_##T(E6) || AXIS_HAS_##T(E7) )
```
individual checks for each E using `AXIS_HAS_##T(Ex)` allows `E0_HARDWARE_SERIAL` to be recognized.

Somehow, the macro `E_AXIS_HAS(T)` is not resolving properly. Should this macro be removed and use the `AXIS_HAS_##T(Ex)` style of checking (like Z axis)?

vivian-ng, 2020-02-22:
@thinkyhead I don't know if this could be an issue with the `RREPEAT` macro not properly expanding the `AXIS_HAS_##T(Ex)` check, since the check uses the macro somewhere further down the line (tracing it through the logic in `macros.h`).

sjasonsmith, 2020-02-23:
@vivian-ng Could you post your actual configuration files?

vivian-ng, 2020-02-23:
[Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/4240606/Config_files.zip)
@sjasonsmith Here are the full configuration files.

Anyway, as mentioned, when I changed `ANY_AXIS_HAS(T)` in `drivers.h` to:
```
#define ANY_AXIS_HAS(T) (    AXIS_HAS_##T(X)  || AXIS_HAS_##T(X2) \
                          || AXIS_HAS_##T(Y)  || AXIS_HAS_##T(Y2) \
                          || AXIS_HAS_##T(Z)  || AXIS_HAS_##T(Z2) \
                          || AXIS_HAS_##T(Z3) || AXIS_HAS_##T(Z4) \
                          || AXIS_HAS_##T(Z3) || AXIS_HAS_##T(Z4) \
                          || AXIS_HAS_##T(E0) || AXIS_HAS_##T(E1) \
                          || AXIS_HAS_##T(E2) || AXIS_HAS_##T(E3) \
                          || AXIS_HAS_##T(E4) || AXIS_HAS_##T(E5) \
                          || AXIS_HAS_##T(E6) || AXIS_HAS_##T(E7) )
```
it could compile, which is why I suspect it has to do with upstream code from this `#define` statement that expands the original `E_AXIS_HAS(T)` part of the `#define`.

vivian-ng, 2020-02-23:
I did a bit of searching through the commits and PR, seems that the switch to use `E_AXIS_HAS(T)`  instead of `AXIS_HAS_##T(Ex)` took place in [PR#16790](https://github.com/MarlinFirmware/Marlin/pull/16790), maybe @Fabi0San can help shed some light on how
```
#define _OR_EAH(N,T)    || AXIS_HAS_##T(E##N)
#define E_AXIS_HAS(T)   (0 RREPEAT2(E_STEPPERS, _OR_EAH, T))
```
works because trying to unwrap/expand those macros is making my head spin due to my poor programming skills.

sjasonsmith, 2020-02-23:
>  trying to unwrap/expand those macros is making my head spin due to my poor programming skills.

It takes a very particular skillset to make sense of this type of preprocessor programming. I'm not very good at it either.

@vivian-ng even though it shows up in the commit with those changes, I'm pretty sure this is a change @thinkyhead had pushed into that PR.

I have been able to determine a couple things:

1. esp32 is not special, it does not work for other platforms either.
2. I don't think it likes repeating over the `defined` statement used when checking for SW serial. I think it is working when checking other conditions, which do not rely on `defined`.
`#define AXIS_HAS_SW_SERIAL(A) ((AXIS_HAS_UART(A) && !defined(A##_HARDWARE_SERIAL)))`

Fabi0San, 2020-02-23:
It seems to me that ANY_AXIS_HAS should include all Es as well.

Maybe @thinkyhead can help shed some light here, these are are part of the changes he added to my CoolStep PR which unfortunately caused some havoc around E(X2, Y2, Z2) axis drivers. Reason for which my PR was mostly reverted by https://github.com/MarlinFirmware/Marlin/commit/18a7276909a3e2aa1761ba0c46577a8f6e7baedb and https://github.com/MarlinFirmware/Marlin/commit/71539bcbeb68284663d0627207ffd5a9a377e920

sjasonsmith, 2020-02-23:
@vivian-ng I posted a pull request that fixes this, though in a different way than just reverting the REPEAT change.

My change is more code than just reverting the change as you did, but I'm hoping it will trigger a bit of discussion regarding the REPEAT macros and their limitations. I see a lot of potential for them to be used incorrectly elsewhere in the future and accidentally cause other issues similar to this.

vivian-ng, 2020-02-24:
Commit from [PR#16940](https://github.com/MarlinFirmware/Marlin/pull/16940) fixed this issue.

Subsequent [commit](https://github.com/MarlinFirmware/Marlin/commit/9a51f29efad894eb96bc86d6bb9b274700d0a816) also works.

closed 2020-02-24 by vivian-ng