MarlinFirmware/Marlin #16916: [BUG] Bugfix-2.0.x Last bugfix does not compile with UBL

Bergerac56, 2020-02-21:
With last bugfix 6320f9795ca91e8c467339afeaa5d3d354500543, and a very basic config file (Just UBL, fixed probe and graphic LCD.), "compile" fails

Compiling .pio\build\megaatmega2560\src\src\feature\binary_protocol.cpp.o
Marlin\src\feature\bedlevel\ubl\ubl.cpp: In static member function 'static void unified_bed_leveling::display_map(int)':
Marlin\src\feature\bedlevel\ubl\ubl.cpp:156:22: error: 'gcode' was not declared in this scope
     const bool was = gcode.set_autoreport_paused(true);
                      ^
Marlin\src\feature\bedlevel\ubl\ubl.cpp:232:5: error: 'set_gcode' was not declared in this scope
     set_gcode.set_autoreport_paused(was);
     ^
*** [.pio\build\megaatmega2560\src\src\feature\bedlevel\ubl\ubl.cpp.o] Error 1

Test config file : 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4235858/Configuration.zip)

ellensp, 2020-02-21:
Yes this is already known, see https://github.com/MarlinFirmware/Marlin/pull/16914/files for a patch.

Bergerac56, 2020-02-21:
@ellensp. Sorry, I was far from marlin for some days. I did not notice that the issue was already found.
And... Thanks. The fix works perfectly. 

I close the issue

closed 2020-02-21 by Bergerac56