MarlinFirmware/Marlin #16919: Crash on endstop

murdock62, 2020-02-22:
### Bug Description

My Min position is 0 on both axis (X and Y).

After homing if i move to 0 (for example G1 X0) then carriage will crash on the endstop.

No problem with 2.0.0

Many thanks for your help.

### My Configurations

[Cfg.zip](https://github.com/MarlinFirmware/Marlin/files/4238708/Cfg.zip)

ellensp, 2020-02-22:
This doesn't make sense, after homing your at X0, Y0 so G1 X0 does nothing. It shouldn't even attempt to move.

murdock62, 2020-02-22:
@ellensp 
> This doesn't make sense, after homing your at X0, Y0 so G1 X0 does nothing. It shouldn't even attempt to move.

oops little precision homing made with bltouch and Z safe homing. 
So the carriage stay to the center of the bed after G28.

After the G28 if i send G1 X0 (or Y0) ---> Crash !

ellensp, 2020-02-22:
"crash" is not a very useful term.... Do you mean that the axis moves towards the end stop and does not stop?

murdock62, 2020-02-22:
Yes quite.  Sorry for the wrong term.  I wanted to find a "synthetic" expression to explain that indeed the carriage did not stop and violently struck the endstops but with my bad English and the late hour, I chose my words wrong.  Sorry!

Marlor, 2020-02-22:
https://github.com/MarlinFirmware/Marlin/issues/16901 there is the reason and a manual fix

murdock62, 2020-02-22:
Thanks @Marlor !

But wow :

Too tired to find the right way to qulifier my problem 
the nose in the middle of the face ... 
(French expression I do not know if it exists elsewhere .. .)

I better go to bed rather than insist on certain evenings ...

Many thanks again !

closed 2020-02-22 by murdock62