MarlinFirmware/Marlin #16920: [BUG] Maybe a problem with 2.0.4.1 release...

Balzac40, 2020-02-22:
Compiling .pio\build\LPC1769\src\src\module\stepper\indirection.cpp.o
Marlin\src\module\stepper.cpp:2567:45: error: macro "BABYSTEP_AXIS" passed 4 arguments, but takes just 3
 2567 |             BABYSTEP_AXIS(X, 0, direction, 0);
      |                                             ^
Marlin\src\module\stepper.cpp:2512: note: macro "BABYSTEP_AXIS" defined here
 2512 |     #define BABYSTEP_AXIS(AXIS, INV, DIR) do{           \
      |
Marlin\src\module\stepper.cpp:2577:59: error: macro "BABYSTEP_AXIS" passed 4 arguments, but takes just 3
 2577 |             BABYSTEP_AXIS(Y, 0, direction, (CORESIGN(1)<0));
      |                                                           ^
Marlin\src\module\stepper.cpp:2512: note: macro "BABYSTEP_AXIS" defined here
 2512 |     #define BABYSTEP_AXIS(AXIS, INV, DIR) do{           \
      |
Marlin\src\module\stepper.cpp:2590:73: error: macro "BABYSTEP_AXIS" passed 4 arguments, but takes just 3
 2590 |           BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z, direction, (CORESIGN(1)<0));
      |                                                                         ^
Marlin\src\module\stepper.cpp:2512: note: macro "BABYSTEP_AXIS" defined here
 2512 |     #define BABYSTEP_AXIS(AXIS, INV, DIR) do{           \
      |
Marlin\src\module\stepper.cpp: In static member function 'static void Stepper::do_babystep(AxisEnum, bool)':
Marlin\src\module\stepper.cpp:2567:13: error: 'BABYSTEP_AXIS' was not declared in this scope
 2567 |             BABYSTEP_AXIS(X, 0, direction, 0);
      |             ^~~~~~~~~~~~~
*** [.pio\build\LPC1769\src\src\module\stepper.cpp.o] Error 1

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Taomyn, 2020-02-22:
Similar for the SK3 mini E3 using VScode

```
Compiling .pio\build\STM32F103RC_bigtree_512K\src\src\module\stepper\trinamic.cpp.o
Marlin\src\module\stepper.cpp:2590:73: error: macro "BABYSTEP_AXIS" passed 4 arguments, but takes just 3
           BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z, direction, (CORESIGN(1)<0));
                                                                         ^
Compiling .pio\build\STM32F103RC_bigtree_512K\src\src\module\temperature.cpp.o
Compiling .pio\build\STM32F103RC_bigtree_512K\src\src\module\tool_change.cpp.o
Compiling .pio\build\STM32F103RC_bigtree_512K\src\src\sd\Sd2Card.cpp.o
Marlin\src\module\stepper.cpp: In static member function 'static void Stepper::do_babystep(AxisEnum, bool)':
Marlin\src\module\stepper.cpp:2590:11: error: 'BABYSTEP_AXIS' was not declared in this scope
           BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z, direction, (CORESIGN(1)<0));
           ^~~~~~~~~~~~~
*** [.pio\build\STM32F103RC_bigtree_512K\src\src\module\stepper.cpp.o] Error 1
```

Moyster, 2020-02-22:
Same for Re-Arm & vscode here.

Interplain, 2020-02-22:
Same Here, using Atom.  SKR 1.3 and TMC2208 drivers fails to build.

ellensp, 2020-02-22:
Default configs (ie BOARD_RAMPS_14_EFB) enabling BABYSTEPING in Configuration_adv.h  and you get this error.. 
So I suspects it is all boards. 
It is also in bugfix 2.0.x

HaxxHaris, 2020-02-22:
Same here, using vscode. SKR 1.4 Turbo and TMC2209 drivers fails to build.

Lord-Quake, 2020-02-22:
I can not confirm the errors you are having.
Compiled `2.0.4.1` successfully with `STM32F103RC_bigtree_512K` board.
Once with `#define BABYSTEPPING` disabled and once enabled.

Post your configs so a closer look is possible.

krille0x7c2, 2020-02-22:
Same here
Marlin/src/module/stepper.cpp: In static member function 'static void Stepper::do_babystep(AxisEnum, bool)':
Marlin/src/module/stepper.cpp:2590:11: error: 'BABYSTEP_AXIS' was not declared in this scope
           BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z, direction, (CORESIGN(1)<0));
           ^~~~~~~~~~~~~
*** [.pio/build/LPC1768/src/src/module/stepper.cpp.o] Error 1

Marlor, 2020-02-22:
Default config + #define BABYSTEPPING
and
CR-10(s) config (no changes)
same error

Taomyn, 2020-02-22:
Attached my configs

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4239446/Configuration.zip)

J0hnMatrix, 2020-02-22:
No issues here too.
I had this error too on my first compile attempt, but after redownloading the files and doing a review on my custom previous 2.0.3 and this new 2.0.4.1, the error is now gone.
First 2.0.4.1 and the new one and don't have found any differences...

Lord-Quake, 2020-02-22:
Need the `platformio.ini` as well.

Taomyn, 2020-02-22:
I'm using the Auto Marlin Builder v2.1.2 extension with VScode so I never touch that. Does it get modified by the extension then?

ChrisBountalis, 2020-02-22:
> No issues here too.
> I had this error too on my first compile attempt, but after redownloading the files and doing a review on my custom previous 2.0.3 and this new 2.0.4.1, the error is now gone.
> First 2.0.4.1 and the new one and don't have found any differences...

Me too, just retried and the error is gone...

Taomyn, 2020-02-22:
I also tried again, in case maybe the extension was also updated, but still fails

Lord-Quake, 2020-02-22:
@Taomyn I used your configs and also edited `platformio.ini` using `default_envs = STM32F103RC_bigtree_512K`
It complied successfully. Give it a try.

Taomyn, 2020-02-22:
And it's ok for me now - all I did was extract the files to a new folder, copied over the config files from the previous folder and hit the build button.

So I compared the folder and found one different file:

[image]

ellensp, 2020-02-22:
+ 20fa62043...c41f3f858 2.0.x      -> origin/2.0.x  (forced update)
Suspect it has  been "quietly" fixed

krille0x7c2, 2020-02-22:
Yea, if you remove that "(coresign(1)<0)) from the BABYSTEP_AXIS function call it will compile. I'll guess it's a human error passing an extra parameter to a function with only three parameters.

Taomyn, 2020-02-22:
Mystery solved - thank-you all

Moyster, 2020-02-22:
bugfix-2.0 still have the problematic commit ( 484e1a624c178f1bb0996892d2c606f6cfebefbe ) , 2.0.4.1 is fine ( 8670df08a299a322f64a7ae259c34c6acd6ef85e )

thisiskeithb, 2020-02-22:
> bugfix-2.0 still have the problematic commit ( [484e1a6](https://github.com/MarlinFirmware/Marlin/commit/484e1a624c178f1bb0996892d2c606f6cfebefbe) ) , 2.0.4.1 is fine ( [8670df0](https://github.com/MarlinFirmware/Marlin/commit/8670df08a299a322f64a7ae259c34c6acd6ef85e) )

That would explain why it’s compiling for some and not others. The change didn’t make its way into `bugfix-2.0.x`.

Scope666, 2020-02-22:
> 
> That would explain why it’s compiling for some and not others. The change didn’t make its way into `bugfix-2.0.x`.

Yeah, 2.0.4.1 just compiled fine for me on LPC1768     (non-bugfix)

PM3Delta, 2020-02-22:
With babysteps enabled, 2.0.4.1 with SKR E3 MINI v1.2.  (default_envs = STM32F103RC_bigtree, BOARD_BTT_SKR_MINI_E3_V1_2) gets the error pointing to the stepper.cpp, where babystep_axis is called with 4 instead of 3 parameters.  When deleting the (coresign...) it compiles ok as thisiskeithb already wrote.

thisiskeithb, 2020-02-23:
This has been patched and [`2.0.4.2`](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.4.2) has been released.

closed 2020-02-23 by thinkyhead

Balzac40, 2020-02-23:
@thinkyhead When I was a C programmer (on last century, yes lol), we used to examine compilation warnings... Maybe the second one is important...

### --- First one ---

In file included from Marlin\src\lcd\menu\menu_temperature.cpp:31:
Marlin\src\lcd\menu\menu.h: In function 'void menu_temperature()':
Marlin\src\lcd\menu\menu.h:343:37: warning: '_thisItemNr' may be used uninitialized in this function [-Wmaybe-uninitialized]
  343 | #define END_SCREEN() } screen_items = _thisItemNr
      |                                     ^
Marlin\src\lcd\menu\menu.h:327:40: note: '_thisItemNr' was declared here
  327 |   int8_t _menuLineNr = encoderTopLine, _thisItemNr; \
      |                                        ^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:340:22: note: in expansion of macro 'SCREEN_OR_MENU_LOOP'
  340 | #define START_MENU() SCREEN_OR_MENU_LOOP(true)
      |                      ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_temperature.cpp:165:3: note: in expansion of macro 'START_MENU'
  165 |   START_MENU();
      |   ^~~~~~~~~~

### --- Second one ---

Marlin\src\sd\SdBaseFile.cpp: In member function 'bool SdBaseFile::open(SdBaseFile*, const uint8_t*, uint8_t)':
Marlin\src\sd\SdBaseFile.cpp:652:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
  652 |       dateTime_(&p->creationDate, &p->creationTime);
      |                 ^~~~~~~~~~~~~~~~
Marlin\src\sd\SdBaseFile.cpp:652:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
Marlin\src\sd\SdBaseFile.cpp:652:35: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
  652 |       dateTime_(&p->creationDate, &p->creationTime);
      |                                   ^~~~~~~~~~~~~~~~
Marlin\src\sd\SdBaseFile.cpp:652:35: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
Marlin\src\sd\SdBaseFile.cpp: In member function 'bool SdBaseFile::sync()':
Marlin\src\sd\SdBaseFile.cpp:1437:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1437 |       dateTime_(&d->lastWriteDate, &d->lastWriteTime);
      |                 ^~~~~~~~~~~~~~~~~
Marlin\src\sd\SdBaseFile.cpp:1437:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
Marlin\src\sd\SdBaseFile.cpp:1437:36: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
 1437 |       dateTime_(&d->lastWriteDate, &d->lastWriteTime);
      |                                    ^~~~~~~~~~~~~~~~~
Marlin\src\sd\SdBaseFile.cpp:1437:36: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]

### --- Third one ---

.pio\libdeps\LPC1769\U8glib-HAL_ID1932\src\clib\u8g_com_i2c.c:44:16: warning: 'u8g_i2c_opt' defined but not used [-Wunused-variable]
   44 | static uint8_t u8g_i2c_opt;   /* U8G_I2C_OPT_NO_ACK, SAM: U8G_I2C_OPT_DEV_1 */
      |                ^~~~~~~~~~~

I hope it helps ; thank you for this really great job and waiting for configurator!

Have a nice day!

Interplain, 2020-02-23:
> 2.0.4.2 release @thinkyhead When I was a C programmer (on last century, yes lol), we used to examine compilation warnings... Maybe the second one is important...
> 
> ### --- First one ---
> In file included from Marlin\src\lcd\menu\menu_temperature.cpp:31:
> Marlin\src\lcd\menu\menu.h: In function 'void menu_temperature()':
> Marlin\src\lcd\menu\menu.h:343:37: warning: '_thisItemNr' may be used uninitialized in this function [-Wmaybe-uninitialized]
> 343 | #define END_SCREEN() } screen_items = _thisItemNr
> | ^
> Marlin\src\lcd\menu\menu.h:327:40: note: '_thisItemNr' was declared here
> 327 | int8_t _menuLineNr = encoderTopLine, _thisItemNr;
> | ^~~~~~~~~~~
> Marlin\src\lcd\menu\menu.h:340:22: note: in expansion of macro 'SCREEN_OR_MENU_LOOP'
> 340 | #define START_MENU() SCREEN_OR_MENU_LOOP(true)
> | ^~~~~~~~~~~~~~~~~~~
> Marlin\src\lcd\menu\menu_temperature.cpp:165:3: note: in expansion of macro 'START_MENU'
> 165 | START_MENU();
> | ^~~~~~~~~~
> 
> ### --- Second one ---
> Marlin\src\sd\SdBaseFile.cpp: In member function 'bool SdBaseFile::open(SdBaseFile*, const uint8_t*, uint8_t)':
> Marlin\src\sd\SdBaseFile.cpp:652:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> 652 | dateTime_(&p->creationDate, &p->creationTime);
> | ^~~~~~~~~~~~~~~~
> Marlin\src\sd\SdBaseFile.cpp:652:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> Marlin\src\sd\SdBaseFile.cpp:652:35: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> 652 | dateTime_(&p->creationDate, &p->creationTime);
> | ^~~~~~~~~~~~~~~~
> Marlin\src\sd\SdBaseFile.cpp:652:35: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> Marlin\src\sd\SdBaseFile.cpp: In member function 'bool SdBaseFile::sync()':
> Marlin\src\sd\SdBaseFile.cpp:1437:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> 1437 | dateTime_(&d->lastWriteDate, &d->lastWriteTime);
> | ^~~~~~~~~~~~~~~~~
> Marlin\src\sd\SdBaseFile.cpp:1437:17: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> Marlin\src\sd\SdBaseFile.cpp:1437:36: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> 1437 | dateTime_(&d->lastWriteDate, &d->lastWriteTime);
> | ^~~~~~~~~~~~~~~~~
> Marlin\src\sd\SdBaseFile.cpp:1437:36: warning: taking address of packed member of 'directoryEntry' may result in an unaligned pointer value [-Waddress-of-packed-member]
> 
> ### --- Third one ---
> .pio\libdeps\LPC1769\U8glib-HAL_ID1932\src\clib\u8g_com_i2c.c:44:16: warning: 'u8g_i2c_opt' defined but not used [-Wunused-variable]
> 44 | static uint8_t u8g_i2c_opt; /* U8G_I2C_OPT_NO_ACK, SAM: U8G_I2C_OPT_DEV_1 */
> | ^~~~~~~~~~~
> 
> I hope it helps ; thank you for this really great job and waiting for configurator!
> 
> Have a nice day!

I see these also when compiling the latest version 2.0.4.1 an have a error message come on my controller Everytime I restart my printer. This message I have to cancel. I think it's something to do with this issue your talking about. I have both TFT 35 V2.0 and TFT 35 V3.0 In original Marlin  it doesn't show, only in Unified menu.

Moyster, 2020-02-23:
> I see these also when compiling the latest version 2.0.4.1 an have a error message come on my controller Everytime I restart my printer. This message I have to cancel. I think it's something to do [...]

did you try another Sdcard / reformating properly your current Sdcard ? as it is now, these (sdcard related) warnings are only showing due to a compiler update, but the code didn't change and should produce the same output once compiled.  
What do you mean by "original Marlin" and what "doesn't show, only in Unified menu" ?  
It sounds like you definitely have an issue, tho, you might want to create an issue with your setup / configs, to make it easier to troubleshoot for devs / maintainers, instead of squatting a (totally unrelated) closed issue ticket :^p

Interplain, 2020-02-23:
> did you try another Sdcard / reformating properly your current Sdcard ? as it is now, these (sdcard related) warnings are only showing due to a compiler update, but the code didn't change and should [...]

When I get a min I will take some pictures of what I mean. Original Marlin screen the one navigated with the encoder. Unified is a menu button version. The TFT 35 V3.0 is able to do both formats. I have few issues an I would love to know what version to use. I have tried 2.0 Bugfix an 2.1 all. I can't get UBL working,  I had Anet A8 working fine with 8bit. But 32 is painful.