MarlinFirmware/Marlin #16923: [BUG] Probing fails on slow probe

Enfostorm, 2020-02-22:
### Bug Description

After updating from an earlier version of marlin (2.0.x in version.h) to 2.0.4.1 the probing on my delta printer seems to have broken. I have reconfigured config.h and config_adv.h from the example files.
I tried to do the auto calibration once I flashed the new firmware, but the probing seems to fail on the slow pass every time with an error message saying Probing Failed.

### My Configurations

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/4239547/Configuration.files.zip)

The part I'd think would influence this behaviour
`#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 5 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE    5 // Z Clearance between multiple probes
//#define Z_AFTER_PROBING            50 // Z position after probing is done

#define Z_PROBE_LOW_POINT          -3 // Farthest distance below the trigger-point to go before stopping`

### Steps to Reproduce

Execute delta auto calibrate, G33 or G33 P1

**Expected behavior:** The calibration finishes

**Actual behavior:** The probe triggers correctly on the fast pass, but fails on the slow pass.

#### Additional Information

[https://youtu.be/xKbENnGO7e0](video)
`SENDING:M111 S255
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
G33 Auto Calibrate
Checking... AC
.Height:350.00  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:134.40
.               Tx:+0.00  Ty:+0.00  Tz:+0.00
reset_bed_level
  current_position= X0.00 Y0.00 Z290.90 : >>> home_delta
  current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, 405.00, [100.00])
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, -5.00, [100.00])
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, 10.00, 10.00)
<<< do_homing_move(X)
delta_endstop_adj:
>>> do_homing_move(X, -0.00, [100.00])
<<< do_homing_move(X)
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, 405.00, [100.00])
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, -5.00, [100.00])
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, 10.00, 10.00)
<<< do_homing_move(Y)
delta_endstop_adj:
>>> do_homing_move(Y, -0.00, [100.00])
<<< do_homing_move(Y)
<<< homeaxis(Y)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, 405.00, [100.00])
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, -5.00, [100.00])
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, 10.00, 10.00)
<<< do_homing_move(Z)
delta_endstop_adj:
>>> do_homing_move(Z, -0.00, [100.00])
<<< do_homing_move(Z)
<<< homeaxis(Z)
>>> set_axis_is_at_home(X)
  current_position= X0.00 Y0.00 Z376.11 :
<<< set_axis_is_at_home(X)
>>> set_axis_is_at_home(Y)
  current_position= X0.00 Y0.00 Z376.11 :
<<< set_axis_is_at_home(Y)
>>> set_axis_is_at_home(Z)
  current_position= X0.00 Y0.00 Z366.11 :
<<< set_axis_is_at_home(Z)
  current_position= X0.00 Y0.00 Z366.11 : sync_plan_position
  current_position= X0.00 Y0.00 Z366.11 : <<< home_delta
>>> Probe::probe_at_point(0.00, 0.00, raise, 0, probe_relative)
  current_position= X0.00 Y0.00 Z366.11 :
>>> do_blocking_move_to X0.00 Y0.00 Z290.90
  destination= X0.00 Y0.00 Z366.11 : destination = current_position
  destination= X0.00 Y0.00 Z290.90 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z290.90 : zone border move
  current_position= X0.00 Y0.00 Z290.90 : xy move
<<< do_blocking_move_to
  current_position= X0.00 Y0.00 Z290.90 : Probe::set_deployed
deploy: 1
Probe::move_z(10.00)
Deploy Z-Probe
echo:busy: paused for user
>>> do_blocking_move_to X0.00 Y0.00 Z290.90
  destination= X0.00 Y0.00 Z290.90 : destination = current_position
  current_position= X0.00 Y0.00 Z290.90 : xy move
<<< do_blocking_move_to
  current_position= X0.00 Y0.00 Z290.90 : >>> Probe::run_z_probe
  current_position= X0.00 Y0.00 Z290.90 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X0.00 Y0.00 Z31.11
  destination= X0.00 Y0.00 Z290.90 : destination = current_position
  current_position= X0.00 Y0.00 Z290.90 : xy move
  destination= X0.00 Y0.00 Z31.11 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z31.11 : z lower move
<<< do_blocking_move_to
  current_position= X0.00 Y0.00 Z100.04 : sync_plan_position
  current_position= X0.00 Y0.00 Z100.04 : <<< Probe::probe_down_to_z
>>> do_blocking_move_to X0.00 Y0.00 Z105.04
  destination= X0.00 Y0.00 Z100.04 : destination = current_position
  destination= X0.00 Y0.00 Z105.04 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z105.04 : z raise move
  current_position= X0.00 Y0.00 Z105.04 : xy move
<<< do_blocking_move_to
  current_position= X0.00 Y0.00 Z105.04 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X0.00 Y0.00 Z11.11
  destination= X0.00 Y0.00 Z105.04 : destination = current_position
  current_position= X0.00 Y0.00 Z105.04 : xy move
  destination= X0.00 Y0.00 Z11.11 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z11.11 : z lower move
<<< do_blocking_move_to
  current_position= X0.00 Y0.00 Z100.22 : sync_plan_position
  current_position= X0.00 Y0.00 Z100.22 : <<< Probe::probe_down_to_z
SLOW Probe fail!
  current_position= X0.00 Y0.00 Z100.22 : <<< run_z_probe
  current_position= X0.00 Y0.00 Z100.22 : Probe::set_deployed
deploy: 0
Probe::move_z(10.00)
Stow Z-Probe
echo:busy: paused for user
>>> do_blocking_move_to X0.00 Y0.00 Z100.22
  destination= X0.00 Y0.00 Z100.22 : destination = current_position
  current_position= X0.00 Y0.00 Z100.22 : xy move
<<< do_blocking_move_to
Error:Probing Failed
Error:Probing Failed

<<< Probe::probe_at_point
Correct delta settings with M665 and M666
>>> do_blocking_move_to X0.00 Y0.00 Z290.90
  destination= X0.00 Y0.00 Z100.22 : destination = current_position
  destination= X0.00 Y0.00 Z290.90 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z290.90 : z raise move
  current_position= X0.00 Y0.00 Z290.90 : xy move
<<< do_blocking_move_to
  current_position= X0.00 Y0.00 Z290.90 : Probe::set_deployed
deploy: 0`

AndersSahlman, 2020-02-23:
Agree, I had this issue before and reported as #16879 I thought it was a bad rebase but now when building on tag 2.0.4.2 I have same problem.

Enfostorm, 2020-02-24:
Just a quick update, it still does not work on 2.0.4.3 or a254354 from bugfix-2.0.x. I have also heard from a couple of other people that were having issues with calibrating their printer.

thinkyhead, 2020-02-24:
Last change to G33 (specifically) was Nov 21 b904ba0f291cdcb9759464f064f85ba066b778ce.
Prior to that the previous change was Sep 29 50e4545255605eb506c20eb107270038b0fe7bdb.

Check out Marlin from before and after these dates to see if they exhibit the issue. And see if you can determine the most recent date where G33 worked. Of course, don't forget to `M502`, `M500` after re-flashing, especially with older versions.

sjasonsmith, 2020-02-24:
G33 is working for me, but I did at one point have to increase the probing distance. You might just need to increase that, or there could possibly be a bug in the probing logic making that necessary.

AndersSahlman, 2020-02-24:
For me this is the commit that breaks it
https://github.com/MarlinFirmware/Marlin/commit/84b6e11bd53acbfde0ad82445e21d7bce1cc14f6
On the commit before that it works and from that it is broken.

AndersSahlman, 2020-02-24:
And yes if you increase the probe distance it will be possible to work , unfortunatly the probing distance need to be at least twice the probe offset and that makes delta calibration very slow if you have a probe with say just over 20mm negative offset as the anycubic kossel probes.

sjasonsmith, 2020-02-24:
Oh wait, maybe I haven't tested it after that change went in, I'm not sure. Disregard my earlier claims. I have an Anycubic Kossel, so I'll hopefully be able to try reproducing this later tonight.

Enfostorm, 2020-02-24:
I have not tried it yet, but I think @AndersSahlman might be right about that commit, M48 the same probing error, so I've tried to take a look at the code myself but had to give up fairly quickly 😇. I did also end up at probe.cpp though, which is the file being modified in that commit. 
I'll try if I can get the previous file to work in 2.0.4.3, I'll post results as soon as I have them.

Enfostorm, 2020-02-24:
Alright, here's what I did:
I have my folder with Marlin 2.0.4.3 downloaded from github. Probing does not work here. I took the probing.cpp and probing.h from the earlier commit 8947622, and just replaced those files in my 2.0.4.3 folder, compiled it, erased EEPROM, and the calibration passed without a hitch.
I did the same, just with probing.cpp and probing.h from commit 84b6e11, and then M48 failed. It has to be something in probing.cpp since probing.h has not been touched.

sjasonsmith, 2020-02-25:
I was able to reproduce the same issues as everyone else (I think). While the commit everyone is pointing to introduced the failure, I don't necessarily think the new code is wrong. It might be fine, but additional changes are still needed to allow G33 to measure the delta height.

I am currently running G33 just fine with that change, but I had to do two things first:

1. Measure the Delta height (from bed to probe) and set it. The Anycubic example uses 320mm, but my printer as assembled is only about 278mm! It doesn't have to be perfect, just close. For me the command was `M665 H278`

2. Increase `Z_PROBE_LOW_POINT` in Configuration.h. This might not be necessary. It depends on how perfectly you measured in the first step. I changed this from -2 to -5 (-5 is already the default in newer examples, but wasn't in my config).

Now it can complete G33!

I think some provision will be needed to allow G33 to measure the height and still continue the probing process. There is a lot of variation in how a user can assemble these printers and endstops, so it is unlikely that examples will actually have perfect heights in them.

Enfostorm, 2020-02-25:
I do not know how to code, but I've been looking at the changes, and my gut tells me there is something wrong with line 620, the one with probe triggered too high:
```
if (probe_down_to_z(z_probe_low_point, MMM_TO_MMS(Z_PROBE_SPEED_SLOW))      // No probe trigger?
        **|| current_position.z > -offset.z + _MAX(Z_CLEARANCE_MULTI_PROBE, 4) / 2  // Probe triggered too high?**
      ) {
```
That one leads to the `SLOW probe fail` message I've been getting as far as I can see.

Also, this exact same bit of code was added to the fast probing in that commit, but now in the main branch it's not there anymore and I can't see in the history when that would have changed.

Enfostorm, 2020-02-25:
The probing works if I comment out line 620 (the probe triggered too high? -line), but I have no idea how to fix that. Maybe @InsanityAutomation knows a bit more about this?
In an unrelated accident I crashed my probe into my bed and the microswitch is buggered, so I won't be able to test anything for a while ;(

InsanityAutomation, 2020-02-26:
My initial code used a variable rather than a fixed number however it was modified prior to merging. Regardless, triggered at 4mm means one heckuva slanted bed.....

```
 if (current_position.z < Z_PROBE_OFFSET_RANGE_MAX)
```

InsanityAutomation, 2020-02-26:
To make the current code match my initial submission,

```
if (probe_down_to_z(z_probe_low_point, MMM_TO_MMS(Z_PROBE_SPEED_SLOW))      // No probe trigger?
        || current_position.z > -offset.z + Z_PROBE_OFFSET_RANGE_MAX  // Probe triggered too high?
      ) {
``` 

I noticed the /2 tagged on the end so the limit was 2mm which is still pretty slanted but within the 3mm I typically fault at...

sjasonsmith, 2020-02-26:
These issues are on delta, so the failures don’t necessarily represent a slanted bed. It ends up failing if the delta height isn’t already very close to the actual value, since the “z=0” point is based on a distance from the Z-max home position.

InsanityAutomation, 2020-02-26:
@sjasonsmith in that case the above, looking at the probe limit, should likely encompass the scenario as that would need to be large enough to accommodate. Is that interpretation correct?

faesali, 2020-02-26:
I have same problem since 2.0.4. My printer corexy and home z max. AUTO_BED_LEVELING_BILINEAR 9point. Probing fails on slow probe at 1st point probe. 2.0.3 works fine.

InsanityAutomation, 2020-02-26:
FYI, Kossell user on discord is reporting that probing is running now where it was failing with 2.0.4 using the change above.

tmyzn, 2020-02-26:
Me and my AKLP face the same problem. (Maybe)

To date, I do not believe that some of the solutions presented are fundamental solutions.
The essence of the problem is that over-reliance on the values ​​written in the configuration may cause the probe to actually fail before it touches the bed.

I think the intent is that if the probe is not working properly for some reason, it will not push the head against the table beyond its limits and break it.
However, it is overturned that the printer will not work due to excessive failsafe.

First, deploy the probe, and then check that the probe works correctly by pressing the microswitch with a finger to check the operation.
Next, the probe is carefully lowered, and it is assumed that the position where the probe touches the bed is the zero point.
In short, actually measure the distance to the bed.
Should actual values ​​be trusted, not set values?

Finally, thanks for Google Translate and You.

thinkyhead, 2020-02-26:
The solution will be to tell the probing function to expect unusual probe values and to return them instead of an error. Apparently `G33` expects probed values to be very far from zero.

thinkyhead, 2020-02-26:
> over-reliance on the values ​​written in the configuration may cause the probe to actually fail before it touches the bed

We have to make some assumption about "what is the current position" in response to a probe. The `G28` probe establishes the zero point, and that point is then used as a reference for all the rest. The expected good result for a probe is "pretty close to zero" based on that initial reference point. When we subtract that initial zero reference, we should get something close to zero at all other points measured with the same probe and the same offset.

If you used an endstop to home Z and you didn't calibrate the probe offset, even though the Z position of the nozzle will be wrong, you can still do a `G29` and all the probe values will come out close to zero.

It's only if you introduce an intentional discrepancy between `G28` and `G29` that you might outsmart yourself into generating errors.

I think the real issue is the way `G33` works: Some of its probing expects to possibly get back way-off values, so it should either be passing the value it expects to the probing function, or it should be passing a flag to tell it not to check the probe result for sanity.

thinkyhead, 2020-02-26:
This should fix the issue with `G33` pretty definitively: cf597e2bb1

InsanityAutomation, 2020-02-26:
For some history of the initial code to fault on high triggers, when a probe had faulted or got stuck on, every point it would raise another <probe clearance> height. On larger machines with over 100 probe points, it would end up going way up and hitting the top (there isnt, and shouldnt be a need for, clamping to zmax here). The infinite raise needed to go away but as delta's are few and far between in developers hands, the G33 case was missed. Based on the value I had put in originally, I believed that was the maximum value allowed for G33 to offset however this has been shown to not be the case.

hamed-ta, 2020-03-26:
I still have this proble on Marline 2.0.5.2

#define Z_CLEARANCE_DEPLOY_PROBE   1 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  1 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     1 // Z Clearance between multiple probes
#define Z_AFTER_PROBING           5 // Z position after probing is done

#define Z_PROBE_LOW_POINT          -5 // Farthest distance below the trigger-point to go before stopping

// For M851 give a range for adjusting the Z probe offset
#define Z_PROBE_OFFSET_RANGE_MIN -30
#define Z_PROBE_OFFSET_RANGE_MAX 30

// Enable the M48 repeatability test to test probe accuracy
#define Z_MIN_PROBE_REPEATABILITY_TEST

// Before deploy/stow pause for user confirmation
#define PAUSE_BEFORE_DEPLOY_STOW
#if ENABLED(PAUSE_BEFORE_DEPLOY_STOW)
  #define PAUSE_PROBE_DEPLOY_WHEN_TRIGGERED // For Manual Deploy Allenkey Probe
#endif

Error:Probing Failed
Probing mesh point 31/36.
Error:Probing Failed
Probing mesh point 32/36.
Error:Probing Failed
Probing mesh point 33/36.
Error:Probing Failed
Probing mesh point 34/36.
Error:Probing Failed
Probing mesh point 35/36.
Error:Probing Failed
Probing mesh point 36/36.
Error:Probing Failed
X:8.00 Y:7.00 Z:5.00 E:0.00 Count X:640 Y:560 Z:2000
ok
M104 S70 T0
M140 S70
ok T:48.17 /70.00 B:46.58 /0.00 @:127 B@:0
ok T:48.14 /70.00 B:46.51 /70.00 @:127 B@:0
G28
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X:0 Y:0 Z:-1200
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X:0 Y:0 Z:2000
ok
M502
echo:Hardcoded Default Settings Loaded
M500
echo:Settings Stored (635 bytes; crc 5797)
M501
echo:V75 stored settings retrieved (635 bytes; crc 5797)
Unified Bed Leveling System v1.01 inactive

Unified Bed Leveling initialized.

UBL reset
G29 P1 V4
Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
Probing around (0.00,0.00).

Probing mesh point 1/36.

Bed X: 8.000 Y: 7.000 Z: 2.020
Probing mesh point 2/36.
Bed X: 8.000 Y: 47.600 Z: 2.063
Probing mesh point 3/36.
Bed X: 49.400 Y: 7.000 Z: 2.133
Probing mesh point 4/36.
[... 1124 lines cut ...]
deploy: 1
Probe::move_z(1.00)
>>> do_blocking_move_to X49.40 Y7.00 Z1.90
<<< do_blocking_move_to
  current_position= X49.40 Y7.00 Z1.90 : >>> Probe::run_z_probe
  current_position= X49.40 Y7.00 Z1.90 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X49.40 Y7.00 Z-10.20
<<< do_blocking_move_to
  current_position= X49.40 Y7.00 Z1.90 : sync_plan_position
  current_position= X49.40 Y7.00 Z1.90 : <<< Probe::probe_down_to_z
SLOW Probe fail!
  current_position= X49.40 Y7.00 Z1.90 : <<< run_z_probe
  current_position= X49.40 Y7.00 Z1.90 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to X49.40 Y7.00 Z1.90
<<< do_blocking_move_to
Error:Probing Failed
<<< Probe::probe_at_point
  current_position= X49.40 Y7.00 Z1.90 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to X49.40 Y7.00 Z5.00
<<< do_blocking_move_to
>>> do_blocking_move_to X8.00 Y7.00 Z5.00
<<< do_blocking_move_to
X:8.00 Y:7.00 Z:5.00 E:0.00 Count X:640 Y:560 Z:2000

hunterk43, 2020-05-24:
I'm also facing this problem. Has anybody found a solution?

boelle, 2020-06-21:
@Enfostorm still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle