MarlinFirmware/Marlin #16925: [BUG] Bed temperature not shown

rs-development, 2020-02-22:
### Bug Description

Since upgrading to 2.0.3 the bed temperature is not shown anymore. See Picture. Quite sure it is Config related, but I could not find the mistake in the git logs.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4239662/Configuration.zip)

#### Additional Information

LPC1768
SKR v1.3
RepRapDiscount Display

[image]

thisiskeithb, 2020-02-22:
Give [2.0.4.1](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.4.1) a try.

rs-development, 2020-02-22:
Sorry, but I would like not to switch again. Last time I switched I got a version that was not even able to compile properly, I think it was 2.0.1.

Any reason why I should switch, something changed?

thisiskeithb, 2020-02-22:
> Any reason why I should switch, something changed?

Your issue may have been patched, so we’d rather you start with the most current version.

ellensp, 2020-02-23:
tested with 2.0.3 and provided Configuration files. I get the same results. Additional notes:
The fan icon is in the active position, even though fan has not been turned on.
The heated bed icon is in the heating position, but bed has not been turned on.
Issue is triggered by something in Configuration_adv.h

ellensp, 2020-02-23:
Issue still persists in latest marlin including bugfix

rs-development, 2020-02-25:
Thank you for testing! I will test next weekend without the combine heater. You wanted to make sure it's not a simple config mistake. I will report back

thinkyhead, 2020-02-26:
Indeed. With `STATUS_COMBINE_HEATERS` the entire top section of the screen is expected to be a single image. But the bed temperature should still be displayed.

thinkyhead, 2020-02-26:
The issue was introduced in 3dc49fd1c6f0be7b765300c433c905c273f36711 and I'll post a solution shortly.

thinkyhead, 2020-02-26:
Test the latest `bugfix-2.0.x` to see if it solves the issue.

rs-development, 2020-02-26:
I think it is related to STATUS_BED_WIDTH, which is zero in this configuration and that's why Bed Temperature is not drawn. I have two non-working printers right now, i am not introducing other issues right now. I will take a look at 3dc49fd later and try to understand what's wrong. Thanks for linking it.

rs-development, 2020-02-26:
https://github.com/MarlinFirmware/Marlin/blob/1955eea1b8f13697259e0e2403d47279d2d1ac6a/Marlin/src/lcd/dogm/dogm_Statusscreen.h#L789

Check this file, I have not tested yet but if the IDE resolves all defines correctly... (did I really think that :dancer: ), this seems the be the issue, not the root cause, but maybe the issue.

rs-development, 2020-02-26:
So a short test with the code below in Line 788, brings the bed temperature back to life!

    #else
      #define STATUS_BED_WIDTH 21
    #endif

boelle, 2020-06-23:
@rs-development  
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

@thinkyhead

rs-development, 2020-07-25:
> @rs-development
> Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.
> 
> @thinkyhead

Thanks for fixing, but I am not willing to invest time into testing at the moment.

closed 2020-08-31 by github-actions[bot]