MarlinFirmware/Marlin #16927: G10/G11 honor M203 E feedrate limits rather than M207 ones

richfelker, 2020-02-22:
I want to be able to dynamically set `M203 E...` feedrate limit to control volumetric extrusion rate, to be able to use the same sliced files with different materials and quickly (during print) experiment with different rates to determine what the materials can handle. However, this is completely incompatible with retraction, which needs much higher rates. I was hoping using firmware retraction with a higher feedrate set via `M207 E...` would ignore the `M203` value and use the `M207` one, but it seems to use the lower of the two, which IMO is not useful for anything.

Is this just a bug that can be fixed and not intentional behavior?

ellensp, 2020-02-23:
M203 sets the machines max limits, you should not be able to override them, other than with new M203 settings

MoellerDi, 2020-02-23:
I'm having similar need to dynamically control volumetric extrusion rate for same reason. IMHO it's not new (see "[FR] Shouldn't firmware retraction ignore max feedrate for E?" #6295).

Maybe something like "default_max_volumetric_extrusion_rate E0 [, E1[, E2...]]" would be required to keep the extrusion rate within proper limits. IMHO an extruder should have two hardware limits; 
"default_max_feedrate" to set proper limits for the stepper motor and
(something like) "default_max_volumetric_extrusion_rate" to set proper limits for the nozzle (to be implemented).

richfelker, 2020-02-23:
For practical purposes there really is no limit on the moto unloadedr; the limit is what it can do under load of pressing the filament through the nozzle. Presently I have `M203 E200` set just for the sake of retraction, which would be horribly wrong if that limit were used for any other purpose. I'd like to be able to get rid of this and have a reasonable `M203` setting, both for controlling extrusion rate to get a good quality print, and for safety against mechanical damage, but without a way to make retraction ignore it I can't.

thinkyhead, 2020-02-26:
It seems sensible that there should be a separate max speed value for E when extruding versus E when moving in reverse.

richfelker, 2020-02-26:
Another data point for current interpretation of `M203` limit being wrong: extruder-axis moves in the LCD menu are *not* using it because it's always set way too high; they're imposing their own, very low, limit for extrusion through the hotend.

And another thing I noticed: config menu doesn't let you set E speed precisely; it steps in units of 1 mm/sec rather than something like 0.01 mm/sec. Setting it down to 1 didn't actually seem to be behaving right either; extrusion was not smooth but pulsing to some extent, possibly a bad interaction with linear advance.

MoellerDi, 2020-06-09:
@richfelker: I was working on PR #17017 to implement such or similar feature. That PR got merged yesterday. So once enabled in configurations file, you can use ```M200 Lx.xx``` to set volumetric limits of your extruders, ```max feedrate``` should match the mechanical upper limits. Limits set via ```M200```won't affect any retraction moves.
Please kindly check if it resolves your FR.

MoellerDi, 2023-06-03:
IMHO VOLUMETRIC_EXTRUDER_LIMIT is not taking effect in E only movements in order to allow fast retraction, load/unload filament, etc. try to actually print something.