MarlinFirmware/Marlin #16929: SKR 1.4 Turbo: 'DGUS_SERIAL_GET_TX_BUFFER_FREE' was not declared

DonaldTheDinosaur, 2020-02-22:
### Bug Description

Replacing Ender 5 Plus stock board with BTT SKR 1.4 Turbo (LPC1769).  Keeping stock DGUS display.

Building Marlin 2.0.4.1 with platformio fails with this error:

> **Marlin/src/lcd/extensible_ui/lib/dgus/DGUSDisplay.cpp:1331:48: error: 'DGUS_SERIAL_GET_TX_BUFFER_FREE' was not declared in this scope**

Note that DGUS_SERIAL_GET_TX_BUFFER_FREE is declared in src/HAL/HAL_AVR/HAL.h and src/HAL/HAL_STM32/HAL.h but not in any other HAL.h files.

### My Configurations

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4240352/Configurations.zip)  My changes are marked with "DTD"

[results.txt](https://github.com/MarlinFirmware/Marlin/files/4240351/results.txt)

### Steps to Reproduce

1. Install my configuration files in 2.0.4.1
2. Build

**Expected behavior:** 

Successful build

**Actual behavior:** [What actually happens]

Build fails with error message given above (and below).  See attached results.txt file.

#### Additional Information

There is a second error:

> **Marlin/src/lcd/extensible_ui/lib/dgus/DGUSDisplayDefinitionOrigin.cpp:238:16: error: 'VP_Fan1_Percentage' was not declared in this scope**

There are also many warnings related to macro expansion.  E.g.:

> **Marlin/src/lcd/extensible_ui/lib/dgus/../../../../inc/../core/macros.h:414:30: note: in expansion of macro 'EVAL1'
>  #define EVAL2(V...)    EVAL1(EVAL1(V))**

These may or may not be related to the DGUS_SERIAL_GET_TX_BUFFER_FREE error.

All errors and warnings are related to the DGUS display.  If you comment out DGUS_LCD_UI_ORIGIN at line 2085 of Configuration.h, the build is successful.

ellensp, 2020-02-23:
This screen type has not been ported to the lpc1768/9

DonaldTheDinosaur, 2020-02-23:
Thanks.

Is it a work in progress?

I see that DGUS_SERIAL_PORT is checked in src/HAL/HAL_LPC1768/HAL.h and it was added to src/HAL/HAL_LPC1768/MarlinSerial.cpp 11 days ago.

Do you know how much remains to be done?

ellensp, 2020-02-23:
Current issue seem to be lpc1768/9 implementation of MarlinSerial doesn't have any GET_TX_BUFFER_FREE type functions at all.

thisiskeithb, 2020-02-23:
[This fork](https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin2.0_Bleeding) from @InsanityAutomation works with the 5 Plus & other Creality printers with dwin LCDs.

DonaldTheDinosaur, 2020-02-24:
@thisiskeithb Thanks.  I'll giving that fork a try.

InsanityAutomation, 2020-02-24:
[image]
Fwiw

InsanityAutomation, 2020-02-24:
Use the creality dwin 2.0 bleeding branch for the skr14 turbo. I'll upload the updated screen files sometime today probably.

DonaldTheDinosaur, 2020-02-24:
@InsanityAutomation.

Thanks!  I've already started playing with the "bleeding" branch.

Have you updated Configuration.h for the SKR 1.4 Turbo?

InsanityAutomation, 2020-02-24:
Yeah, uploaded more tweaks this morning

DonaldTheDinosaur, 2020-02-24:
That's great!  I was just starting to hack on the version I downloaded yesterday.  It's much better to have your implementation to start with so I don't have to change later.

DonaldTheDinosaur, 2020-02-24:
@InsanityAutomation 

I pulled your update and set these values in Configuration.h:

> #define MachineEnder5Plus	// DTD
> #define BedDC		// DTD
> #define SKR14Turbo	// DTD
> #define SKR_2209	// DTD
> #define SKR_UART	// DTD

I get successful compilation but with warnings in Creality_DWIN.cpp and SdBaseFile.cpp.

In Creality_DWIN.cpp, there is no default condition for the chained "if" starting at line 1309 in function RTS_HandleData().   This would result in "min", "max", and "axis" being used without having been set if recdat.data does not have one of the expected values.

In SdBaseFile.cpp, these warnings occur at 652 and 1437:

> taking address of packed member of 'directoryEntry' may result in an unaligned pointer value

Fax2u, 2020-04-05:
> @InsanityAutomation
> 
> I pulled your update and set these values in Configuration.h:
> 
> > #define MachineEnder5Plus	// DTD
> > #define BedDC		// DTD
> > #define SKR14Turbo	// DTD
> > #define SKR_2209	// DTD
> > #define SKR_UART	// DTD
> 
> I get successful compilation but with warnings in Creality_DWIN.cpp and SdBaseFile.cpp.
> 
> In Creality_DWIN.cpp, there is no default condition for the chained "if" starting at line 1309 in function RTS_HandleData(). This would result in "min", "max", and "axis" being used without having been set if recdat.data does not have one of the expected values.
> 
> In SdBaseFile.cpp, these warnings occur at 652 and 1437:
> 
> > taking address of packed member of 'directoryEntry' may result in an unaligned pointer value
> 
> [EDIT] To be clear, this is not intended as criticism in any way. I'm just making you aware of my observations. I'm very grateful for your work.

I did the same on a CR-10S pro and couldn't get the DGUS screen to function properly.
also the SKR 1,4 now shows up as a card reader without a card. (serial ports are -1 and 0)

Any ideas for a fix would be greatly appreciated.

boelle, 2020-06-20:
@DonaldTheDinosaur is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle