MarlinFirmware/Marlin #16931: Fysteclcd dgus with mks 1.4  problem

manarsak, 2020-02-23:
Hi, I have a mks1. 4 board and I want to connect a fystec lcd screen when downloading the program. I see this message. I have a problem, I have another question. What are the appropriate settings for the screen to work and what are the appropriate pins on the board to connect it to the screen

[image]
[image]

ellensp, 2020-02-23:
No config files, not text, no versions, practically no information provided.  An image may be worth a 1000 words, but we really need the 1000 words and config files!

manarsak, 2020-02-23:
I'm sorry, these are the configuration files
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4241384/Marlin.zip)

ellensp, 2020-02-24:
Confirmed. This does not compile. Compiles on 2.0.0 and 2.0.1 but anything newer than 2.0.1 that you get the "undefined reference to `ExtUI::FileList::FileList()'"

ellensp, 2020-02-24:
Seems to have stopped working on commit 0e6ef65aec52980daeda09d7a6e8cd98b97f3806

manarsak, 2020-02-24:
My friend, I know what the message means, but what is the reason for this problem, knowing that when you comment on the option dgus fystec, things are okay

//#define DGUS_LCD_UI_FYSETC

boelle, 2020-06-23:
@manarsak 
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

thinkyhead, 2020-06-25:
The latest code builds fine with your configs.

closed 2020-06-25 by thinkyhead