MarlinFirmware/Marlin #16933: onprintingFinished() and onPrintingStarted() callbacks/signals [FR]

reddomon, 2020-02-23:
I am developing  a nextion UI. Make some callback changes but I think marlin needs some sort of callbacks to manage extended UI's.

### Description
Extensible UI development
### Feature Workflow

1. When start a job onPrintingStarted() callback is called
2. Then prints normally
3. When job is finished, onPrintingFinished() callback is called.

#### Additional Information
[image]
[image]
[image]
[image]

[image]
[image]

marciot, 2020-02-24:
@reddomon: I developed the ExtensibleUI for LulzBot and the reason there isn't start/stop callbacks is that in the case of tethered printing, "start of print" is an entirely abstract concept. Marlin does not have anyway of distinguishing between interactive commands being sent from the host and an actual print, nor does it really have anyway of knowing when the print has ended.

That said, there is a good proxy for the functionality you desire and that is the print timer. So long as you arrange it so the slicer calls M75 and M77 (such as in your start and end GCODE), then there is an clear way of knowing when a print is in progress in either tethered or untethered mode. This is how it is done in the LulzBot TAZ Pro and there *are* callbacks for those actions:

```
  void onPrintTimerStarted();
  void onPrintTimerPaused();
  void onPrintTimerStopped();
```

reddomon, 2020-02-24:
Thank you @marciot clearifying that for me :). I see. I thought that if Marlin can call a function when print is done like serial out "print is done", maybe we could do the same thing for UI. My intend is using Marlin as a laser driver, so i think  I can use OnPrint.Timer...() calbacks, adding M75 and M77 command into GCODE. Thank you again.

closed 2020-02-24 by thinkyhead

reddomon, 2020-04-24:
Hi there. I dont know is it wrong or right but i resolved the problem (may not be a problem) this way,I added  a function calls after "SD start reading" and an another function call after "SD finish reading".

thinkyhead, 2020-04-26:
If you have laser questions you should chat with @shitcreek who is also doing laser work with Marlin.