MarlinFirmware/Marlin #16934: [BUG] 2.0.4 && 2.0.4.2 Heatbed stays on after Print finished, no Reaction for "M140 S0"

Flutschi84, 2020-02-23:
### Bug Description

After a Print from the SD Card, the Heatbed stays at Temperature. 
The printer then does not respond to any further host or menu commands. Not even on M140 S0. Only a reset brings the printer back to normal behavior. 
 
When printing through the host, everything behaves normally.

### My Configurations
My conf's:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4240923/Configuration.zip)

In the knob.gco file you can see my Print-End G-Code performs a M140 S0 for shutting down the bed (and something else).
[knob.zip](https://github.com/MarlinFirmware/Marlin/files/4240929/knob.zip)

### Steps to Reproduce

1. Put the SD Card with the *.gco file into the Printer and start the Job from SD Card
2. Print behaves normally
3. When Job comes to the end, this gcodes are performed: (See the knob.gco file)
G92 E0
G1 E-5 F2100
G27 P2
M107
M104 S0
M140 S0
M84
M73 P100 R0

**Expected behavior:** All gcodes will be executed, including S140 S0, so the bed is turned off

**Actual behavior:** 
**- The Heatbed stays on,** 
- The other G-Codes from the files end are executed normally. Means It goes to Parkposition, Fan, NozzleHeat and steppers are disabled, DisplayPrograss is at 100%.
- No Command from Host or DisplayMenu will be accepted 
- Only Reset brings back normal behavior 

#### Additional Information
Printer has neopixel and these are turning to green after print.

Using MKS SGEN_L Board @ Marlin 2.0.4.2 Release
Same Behavior @ 2.0.4 Release
Last good @ 2.0.3 Release

Moyster, 2020-02-23:
was solved in the latest bugfix : https://github.com/MarlinFirmware/Marlin/commit/7bf3581b024c2942d80d03d7e7abeccca612587f  

Too bad it didn't make it in the 2.0.4.1 release.

Flutschi84, 2020-02-23:
Ok, i will try and report.

Thanks

Flutschi84, 2020-02-23:
I can confirm that with latest 2.0.x Bugfix the beavior of not shutting down the bed went normal.

closed 2020-02-23 by Flutschi84