MarlinFirmware/Marlin #16936: [BUG] "Error:SD read Error" at end of print, and no green printer event LED

Flutschi84, 2020-02-23:
### Bug Description

With latest Bugfix, at the End of SD Printing  i see an "Error:SD read Error" in my Terminal (Repetier Host @ Repetier Server) and the LEDs stays white. (Printer event LEDs are Neopixel)

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4241300/Configuration.zip)

[Example gco.zip](https://github.com/MarlinFirmware/Marlin/files/4241301/Example.gco.zip)

### Steps to Reproduce

1. Start Printing *.gco File from SD
2. Wait till end
3. The End-Gcode is performed well

**Expected behavior:**
Printer Event LED should change to green

**Actual behavior:**
- LED stays white
- I can see "Error:SD read Error" in Terminal@Host
- All SD menus disapear in LCD Menu

#### Additional Information
using Latest bugfix 2.0.x on MKS SGEN_L
has defined the LCD reader instead of the onboard Reader. (onboard Reader only used by bootloader for FirmwareStuff)
Worked for me with Marlin 2.0.3 rel and with 2.0.4.2 rel.

Flutschi84, 2020-02-23:
Turns out that both test Prints did stop with the Same Error Message (see above) at different Times.
But both near to the end of the Print. 
confusing that the "End of print" G-Codes were performed, but this could be due to the hard-coded values in configuration_adv.h

At the moment i'm testing with different SD Card.....shame on me if thats the reason....

i will report

Flutschi84, 2020-02-23:
OK, different SD Card , same Problem,
Print finishes correcltly with all the end gcode, goes to park position, set display progress to 100%, and then comes the "SD-Card read Error", SD Menu disapear, LED stays white

Moyster, 2020-02-23:
Right now, I think bugfix matches 2.0.4.2 so you might have an issue with your bugfix branch.  
Can you check if this change was applied to your bugfix : https://github.com/MarlinFirmware/Marlin/commit/7bf3581b024c2942d80d03d7e7abeccca612587f ?  

It's wierd that it works on 2.0.3 & 2.0.4.2 but not bugfix.

Flutschi84, 2020-02-23:
> Can you check if this change was applied to your bugfix : 7bf3581

I will try but for the Moment i'm drunk cause of carnevallitititisssity
..... so tomorrow i will give it a try...  Thanks🥳

Flutschi84, 2020-02-24:
I have checked, all three files contains the changes from [29ec868](https://github.com/MarlinFirmware/Marlin/commit/29ec868c4bc19058f02039c2a90e21fed816c257)

Moyster, 2020-02-24:
> I have checked, all three files contains the changes from [29ec868](https://github.com/MarlinFirmware/Marlin/commit/29ec868c4bc19058f02039c2a90e21fed816c257)

Nah, my bad I didn't word it properly : https://github.com/MarlinFirmware/Marlin/commit/29ec868c4bc19058f02039c2a90e21fed816c257 in bugfix introduces changes that are causing the error you encouter, which is why bugfix doesn't work for you but previous releases do.

Flutschi84, 2020-02-24:
> 29ec868 in bugfix introduces changes that are causing the error you encouter, which is why bugfix doesn't work for you but previous releases do.

You are right, i tested with previous Release 2.0.4.2 and there was no problem with SD read error.
But i needed to switch to bugfix branch, because of other Problems(Bed was no disabled after print).
Now i tested the new 2.0.4.3 rel  that contains the "Commit last SD line before fileHasFinished"-Feature and for me the problem moved from Bugfix to Release Branch.

Am i doing something wrong in my config or hardware? Or have others the same problem?

thinkyhead, 2020-02-24:
```diff
-      if (n < 0) { SERIAL_ERROR_MSG(MSG_SD_ERR_READ); continue; }
+      if (n < 0 && !card_eof) { SERIAL_ERROR_MSG(MSG_SD_ERR_READ); continue; }
```

closed 2020-02-24 by thinkyhead