MarlinFirmware/Marlin #16941: [BUG] Babysteppeing Not moving in 2.0.4.1 and also latest Bugfix

sismo99, 2020-02-23:
###Bug Description

After updating to Latest Marlin 2.0.4.1 and later to the latest bugfix (23-02-2020) when i try to adjust my z offset with the babytep function the nozzle does not move to any direction (up or down) the values goes up or down but the nozzle does not move

when not printig i can go to the move axis menu and it does move well but not in the babysteping function, also y can adjust the z offset with the M851 command (but i have to guess it, print a proof and adjust it like that) 

i tried and lf build of marlin 2.0.3 (2 Weeks Ago) and worked  well so i guess the priblem is in this latest build

###My Configurations

Ender-3 Pro
SKR Mini E3 1.2
Latest Marlin Bugfix (23-02-2020)
EZABL system

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/4242417/Configuration.Files.zip)

### Steps to Reproduce

1. Update To The lastest stable or bugfix  Marlin 
2. Try to configure your Z-Offset 
3. Start a print and double clic in the dial button
4. Move the dial to any direction you can se the values change but not the nozzle

**Expected behavior:**  The nozzle move a little up or down

**Actual behavior:** It does not move nothing

#### Additional Information

As i say in the move axis menu it moves but not while you are in babystepping

thisiskeithb, 2020-02-23:
It was working after I had reported it initially in https://github.com/MarlinFirmware/Marlin/issues/16941, but it looks like babystepping is broken again using 2bcc2ec ☹️

goofball222, 2020-02-24:
Seeing the same thing here with 2bcc2ec freshly compiled. Back to M211 S0 and negative G1 Z axis moves I go.

brejzek, 2020-02-24:
I have the same problem ender 3 pro skr 1.3. Which version to roll back to?

XBrav, 2020-02-24:
> I have the same problem ender 3 pro skr 1.3. Which version to roll back to?

Marlin 2.0.3 is working fine for me. It's definitely something in 2.0.4 forward.

thisiskeithb, 2020-02-24:
> Marlin 2.0.3 is working fine for me. It's definitely something in 2.0.4 forward.

There was a bug in `2.0.3` that prevented extruder motors being turned off and also would leave the hotend on stuck on your print near the end if you have status LEDs, so be careful with it.

Kraken3d, 2020-02-24:
I also concur. Babysteps are not functional in the latest build. I just finished recompiling my firmware from the previous release and babysteps are no longer working. Just as the original poster had mentioned.

SCiunczyk, 2020-02-24:
I have this same problem on megaatmega2560/RAMPS/TMC2130

Minims, 2020-02-24:
Same here on SKR Pro with the last bugfix-2.0.x.

thinkyhead, 2020-02-24:
I have it working on an LPC1769 now. I'll push a fix soon after I check out the reason why it works.

SCiunczyk, 2020-02-24:
Great, I'll test in on my hardware immediately.

thinkyhead, 2020-02-24:
Cool. The patch is now pushed to `bugfix-2.0.x`. The problem was my misunderstanding of how I set up the babystepping array. I added a "hidden" option at the top of `stepper.cpp`. Comment out `BABYSTEPPING_EXTRA_DIR_WAIT` to leave out the extra delays around the last DIR pin change in a babystep. This can make babystepping smoother but less delay might trigger errors with certain Trinamic drivers.

SCiunczyk, 2020-02-24:
I can confirm, it's working like a charm on my megaatmega2560/RAMPS/TMC2130. Many thanks @thinkyhead !

Roonster78, 2020-02-24:
I can confirm also on BTT SKR mini E3 V1.2/TMC2209! Thanks @thinkyhead

Powerprobot, 2020-02-24:
I can confirm working on BTT SKR1.3 with TMC2208 Uart! great work! Thanks @thinkyhead

thinkyhead, 2020-02-24:
Good to hear it's working. I did some additional things last week to try to mitigate bad timing shutting down TMC2209 drivers (particularly on SKR boards) so hopefully now that the calls are working again we'll get a chance to do more testing and patching of that issue.

closed 2020-02-24 by thinkyhead

sismo99, 2020-02-25:
i just have the chanceto cimpile the new bigfix and just to confirm that it is working  great (i know that it is closed but as posted this i want to confir  that it was fixed thanks a lot Thinkyheadkeep with the great work

bill-orange, 2020-02-28:
Still appears broken on Melzi board.

larsnordstrom, 2020-03-11:
I have the same issue on 2.0.4.4 with Einsy Rambo + PINDA2.

thinkyhead, 2020-03-11:
See if `BABYSTEPPING_EXTRA_DIR_WAIT` helps at all. And then see if `INTEGRATED_BABYSTEPPING` helps at all.