MarlinFirmware/Marlin #16942: [BUG] Z axis noise

lamosca01, 2020-02-24:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

swilkens, 2020-02-25:
So I think I understand what you mean.

With the latest configuration files, a new feature was turned on by default called 'HYBRID_THRESHOLD'

By default, this sets the threshold for the Z axis at 3 mm/s, which is basically immediately.

For a quick fix, do:

`M913 Z50`

lamosca01, 2020-02-26:
Thank you! Just that and voila. Regards!

closed 2020-02-26 by thinkyhead