MarlinFirmware/Marlin #16946: [BUG] BLTOUCH_HS_MODE does not raise Z

marciot, 2020-02-24:
### Bug Description

With BLTOUCH_HS_MODE enabled, the BLTouch enters an error state without any motion happening on the Z axis.

### My Configurations

[Marlin Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4246121/Marlin.Configs.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Enable BLTOUCH_HS_MODE
2. Run "G29 P1"

**Expected behavior:**

The Z axis should lift immediately after the BLTouch makes contact with the bed.

**Actual behavior:**

The BLTouch makes contact with the bed, but no motion happens on the Z axis and the BLTouch enters an error state.

thinkyhead, 2020-02-26:
Which of the differences introduced by `BLTOUCH_HS_MODE` being turned on seems the most likely culprit? I notice there are very few places where it is mentioned.

marciot, 2020-03-02:
@thinkyhead: I didn't get a chance to investigate it fully, I just noticed it wasn't doing what I expected so thought I would make record of it.

boelle, 2020-06-20:
@marciot is the isuue gone?

marciot, 2020-06-21:
@boelle: I do not know whether it is gone, but I no longer need it fixed. Closing.

closed 2020-06-21 by marciot