MarlinFirmware/Marlin #16949: [FR] Higher jerk at the beginning of travel moves

josedpedroso, 2020-02-24:
### Description

I'm using Marlin 1.1.9, and have no plans to switch to 2.0 for now, because that would imply losing features due to larger binaries. I understand that 2.0 now uses junction deviation as default, and it's available in 1.1.9 too, but I'm not using it because of space constraints, though perhaps the same general idea can be applied there too.

I like to print with low jerk, high acceleration, and high top speed, because that allows my printer to do well on fine details without sacrificing speed on larger parts. However, there's a drawback. Higher jerk helps reduce stringing a lot, at least on PLA. My retraction distance is unusually large because of that. So I had the idea of using different X/Y jerk settings for travel and printing moves.

### Feature Workflow

Added next to jerk settings in Configuration.h:

`#define EXTRA_TRAVEL_XYJERK           10.0 //additional jerk at the beginning of non-printing moves`

Changed in planner.cpp, inside Planner::_populate_block(), after   #else // Classic Jerk Limiting:

```
    LOOP_NUM_AXIS(i) {
      const float jerk = ABS(current_speed[i]);   // cs : Starting from zero, change in speed for this axis
      float maxj = max_jerk[i];                   // mj : The max jerk setting for this axis
      #ifdef EXTRA_TRAVEL_XYJERK
      	if ((de <= 0) && ((i == X_AXIS) || (i == Y_AXIS)))
      	  maxj += EXTRA_TRAVEL_XYJERK;
      #endif
      
      if (jerk > maxj) {
```

...

```
              ( (v_entry < 0 || v_exit > 0) ? (v_entry - v_exit) : MAX(-v_exit, v_entry) );

        float maxj = max_jerk[axis];
        #ifdef EXTRA_TRAVEL_XYJERK
          if ((de <= 0) && ((axis == X_AXIS) || (axis == Y_AXIS)))
            maxj += EXTRA_TRAVEL_XYJERK;
        #endif
        if (jerk > maxj) {
          v_factor *= maxj / jerk;
          ++limited;
        }
```

#### Additional Information

I've done some short tests with the same g-code for [this model](https://www.thingiverse.com/thing:2219103) before and after, and this does seem to reduce stringing a lot with low jerk. I was able to reduce my usual retraction distance without apparent loss in quality, and the prints probably go a bit faster too. More testing is needed, but so far it looks good to me.

However, the code in planner.cpp is quite involved and I'm not sure this actually does what I think it does.

josedpedroso, 2020-02-25:
I've gradually increased EXTRA_TRAVEL_XYJERK to 32 and the printer seems to handle it fine. I've also reduced retraction further with no loss in quality.

thinkyhead, 2020-02-25:
Out of curiosity, with just the same features as 1.1.9, how much larger was your build of 2.0?

thinkyhead, 2020-02-25:
Having different jerk for travel seems sensible. We have different accelerations for travel versus print moves, and jerk is the low limit for that. But note that at the current time you can actually use the E jerk as a good proxy since it only applies when doing print moves. If you set it lower, then it will start to kick in, whereas typically we set E jerk high so it won't get in the way.

josedpedroso, 2020-02-25:
> 
> 
> Out of curiosity, with just the same features as 1.1.9, how much larger was your build of 2.0?

I didn't try it. I just read in a few places that Marlin 2.0 is larger and I'm already at the limit with 1.1.9. I've only got an 8 bit board, so it probably doesn't make sense to switch now.

> But note that at the current time you can actually use the E jerk as a good proxy since it only applies when doing print moves.

I've actually reduced my e-jerk from the default 5 to 4, but I fear that reducing it further could make retractions less effective.

thinkyhead, 2020-02-25:
I'm all for the addition, and it's a pretty quick change. I'll post something to test later.

thinkyhead, 2020-02-26:
> I've only got an 8 bit board, so it probably doesn't make sense to switch now.

Marlin 2.0 is better for 8 bit boards, as well. Many optimizations over 1.1.

josedpedroso, 2020-02-26:
I might try it, then. Thanks.

closed 2020-02-26 by thinkyhead

josedpedroso, 2020-02-26:
Hey, my understanding of the planner is a bit spotty. Could you explain if this really only applies to the beginning of travel moves, at least in the usual case of travels existing only between prints? And why is the 2nd block of changes in the planner not necessary? Thanks.

thinkyhead, 2020-02-26:
> Could you explain if this really only applies to the beginning of travel moves, at least in the usual case of travels existing only between printing moves?

- The jerk adjustment will apply to all direction changes, not just the start of moves from a standstill.

> why is the 2nd block of changes in the planner not necessary?

- It was. I jumped the gun on pushing to bugfix.

josedpedroso, 2020-02-26:
I guess my question is, then, how is jerk applied to the end of a travel move without knowing in advance the speed and direction of the next move?

My admittedly rough understanding of the planner is that when a block is filled in, it always ends with a deceleration and jerk to a halt, but as the next block is filled in, the actual acceleration and jerk of that new move is used to make the transition, discarding the halt. If that's right, with this change, issuing a travel move and then a printing move would effectively only use increased jerk at the beginning of the travel, which is what I actually intended to do.

To be clear, I'm happy with the result I got regardless of that detail, but the idea was to increase jerk where stringing actually begins to happen and avoid excessive vibration when a printing move begins. Given the good results I got, I wonder if it's even worth pursuing that last detail, but I'm curious about it anyway.

thinkyhead, 2020-02-26:
> how is jerk applied to the end of a travel move

It applies to instantaneous changes in speed and direction and determines the speed at which moves are joined.

josedpedroso, 2020-02-26:
I understand that. I'll try to pose my question more clearly: in the transition from a travel move that had extra XY jerk applied, to a printing move that will only have the regular jerk, which jerk value will actually be used?

From what I gathered, only the regular, non-incremented jerk value will be used in that transition.

thinkyhead, 2020-02-26:
Jerk is applied based on the move being added and pertains to the junction between the previous move and the new one.

josedpedroso, 2020-02-26:
Well, the end result of that is this extra XY jerk effectively only being applied at the beginning of a travel, if the next move prints. Thanks.

thinkyhead, 2020-02-26:
> extra XY jerk effectively only being applied at the beginning of a travel, if the next move prints

No. It will be applied right away on the first segment which is a travel move, and it will continue to apply in the joining of all the travel segments that follow. The jerk applied on the travel-to-printing transition —and the joining of all the travel segments that follow— will be the normal XY jerk.

josedpedroso, 2020-02-26:
I wasn't taking into account the case of having multiple travel segments in sequence. In any case, we're in agreement about the travel-to-printing transition, which is the one that really matters for print quality.