MarlinFirmware/Marlin #16950: [FR] Rearrange probe order for UBL so it minimizes travel

marciot, 2020-02-24:
It would be nice to be able to restore a sequential probe sequence in UBL, like when using bilinear bed leveling. Right now it jumps from point to point in a manner that involves a lot of travel. It simply doesn't look very good. I would be curious to know why UBL orders the probe points in such a peculiar way.

ManuelMcLure, 2020-02-24:
UBL tries to probe in concentric circles starting at wherever the probe was when the probing started (usually the center of the bed when ``Z_SAFE_HOMING`` is in use).
I believe the reasoning is to keep the nozzle from ramming the bed hard if the bed is very unlevel (i.e. tilted) - however this is probably more important when doing manual probing that when doing automated probing.

ManuelMcLure, 2020-02-24:
Or, per the code comments:
```
 283    *   When you do a G28 and G29 P1 to automatically build your first mesh, you are going to notice that
 284    *   UBL probes points increasingly further from the starting location. (The starting location defaults
 285    *   to the center of the bed.) In contrast, ABL and MBL follow a zigzag pattern. The spiral pattern is
 286    *   especially better for Delta printers, since it populates the center of the mesh first, allowing for
 287    *   a quicker test print to verify settings. You don't need to populate the entire mesh to use it.
 288    *   After all, you don't want to spend a lot of time generating a mesh only to realize the resolution
 289    *   or probe offsets are incorrect. Mesh-generation gathers points starting closest to the nozzle unless
 290    *   an (X,Y) coordinate pair is given.
```

DroneMang, 2020-02-25:
I would second this.  I have my z safe homing point very near where x and y home to so traveling to the center on a 500x500 bed is unnecessary.

Also, along with this the G26 test pattern seems to follow the same movement as the probing, and the nozzle loses its prime on the long moves and does not therefore show the first layer height well.

Otherwise UBL is really really great!

marciot, 2020-03-02:
> UBL tries to probe in concentric circles starting at wherever the probe was when the probing started (usually the center of the bed when Z_SAFE_HOMING is in use).

I think there might be a middle ground here. Maybe rather than probing points in increasing distance from the origin, which causes the north, west, south and east points to be probed first, followed by the northeast, northwest, southwest, and southeast points; instead pick a starting point on the grid closest to the origin, then do a square spiral "moving" from that center point. hitting the center point, then north, NW, west, SW, south, SE, east, NE... etc. This will lead to a pattern which is more visually pleasing and with half the travel.

thisiskeithb, 2021-04-30:
#21387 was merged. Closing.

closed 2021-04-30 by thisiskeithb