MarlinFirmware/Marlin #16951: [BUG] No Media with inserted SD Card in Menu (MKS_GEN_L+REPRAP Discount Display)

Locus222, 2020-02-24:
**Description**
The SD card reader does initialize when inserting SD card, but menu constantly shows "No Media". Tested with 2xSDHC(both FAT32 and 16GB, one fresh formated) and 1x microSHC(with adapter). All got detected, none got listed in menu. SD Slot is on display board. (display is used from old tevo tarantula and i had to rotate the plugs 180deg. Cuttting the orientation markers on the plugs to get them into the sockets).

Marlin 2.0.4.2
enabled: #define REPRAP_DISCOUNT_SMAR
(https://github.com/MarlinFirmware/Marlin/files/4246620/debug.pin.layout.txt)
T_CONTROLLER
enabled #define SDSUPPORT

**Steps to Reproduce:**

1. MKS_GEN_L
2.Default configuration.h and configuration_adv.h
3.Enable #define SDSUPPORT in configuration.h

**Expected behavior:** 
Have the option to select "Print form SD"

**Actual behavior:** Menu option not changed. constant "No Media" message regardless the presence of the SD card in the slot

#### Additional Information
video of the behavior --> https://www.youtube.com/watch?v=-UbTaDt3aiE 
[debug pin layout.txt](https://github.com/MarlinFirmware/Marlin/files/4246630/debug.pin.layout.txt)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4246626/Marlin.zip)

hamster65, 2020-02-25:
I might have the same problem with my MKS_SGEN_L board and full graphic reprap discount display.
OT: Is it possible to flash the firmware using the card reader of the lcd?

Locus222, 2020-02-25:
I have found that configuration_adv.h contains section for sd support -> line 1013 - #if ENABLED(SDSUPPORT).

a bit lower there is  #if HAS_SDCARD_CONNECTION which contains(in default):
 //#define SDCARD_CONNECTION LCD 

regardless of commenting or uncommenting the line it was not working. It seems to me that the HAS_SDCARD_CONNECTION is not being applied to REPRAP Discount Display (Smart in my case, Full in yours). But i am not able to check it deeper.

TheNeskik, 2020-02-25:
I don't have this issue with e6a7be9
I'll look deeper later with the latest bugfix.

thinkyhead, 2020-02-25:
What was the last version of Marlin where this worked?

TheNeskik, 2020-02-25:
@thinkyhead 2.0.3 is working fine for me with MKS SGEN L + REP RAP DISCOUNT SMART,
2.0.3 is working fine with MKS GEN L.

I will look in a few hour with 2.0.4.x unless you found something before me.

thinkyhead, 2020-02-25:
@Locus222 — The original issue concerns a RAMPS-like GEN-L and for that board `HAS_SDCARD_CONNECTION` is not relevant.

Locus222, 2020-02-25:
> @Locus222 — The original issue concerns a RAMPS-like GEN-L and for that board `HAS_SDCARD_CONNECTION` is not relevant.

ok, as i have MKS GEN L, therefore the `HAS_SDCARD_CONNECTION` should be relevant for me?

last version of the Marlin i used with the RR-SMART-Display where it was working in version v1.1.x

are there some requirements for the cards? I assume FAT32, are there any allocation unit size or max  card size?

TheNeskik, 2020-02-25:
@Locus222 The fact is that the MKS GEN L didn't have a sd port on board, you have to use the bus on your LCD. **HAS_SDCARD_CONNECTION** let you choose which port to use (board or LCD in most of cases). That's why it is not relevant in your case.

All my SD cards are FAT32.

Maybe try to use the latest bugfix and init EEPROM ? (don't think this will solve your issue)
I'll try to test with my MKS GEN L 

@hamster65 My MKS **S**GEN L works like a charm with latest bugfix. I never manage to flash the firmware with the card reader on LCD.

thinkyhead, 2020-02-26:
The concept introduced in 290466578f8bb6ba0cd61e1dd3dc0839ae53d4f3 is increasingly obsolete as more platforms have a variety of options. Maybe the flag will be added to pins files that define `SDCARD_CONNECTION`, but for now we'll leave it open to be modified, and boards that don't use it will just ignore it.

closed 2020-02-26 by thinkyhead

Locus222, 2020-02-28:
faulty cable has been identified. All works perfectly with  version 2.0.4.3. 

thank you very much for your effort.

leytanelectronics2, 2020-05-11:
im using the ramps 1.4 board with an arduino mega 2560, is there a solution for this setup?