MarlinFirmware/Marlin #16952: [FR] Manual and easier filament switch routine

PuddingBaer91, 2020-02-24:
### Description

Because i don't like the automatic filament switch routine and also to keep it easier it would be nice to add a simple filament switch routine where only the print head is lifted and e-stepper is disabled to manually feed out and in filament. There is also a second easier automatic method explained.

For me it is much easier because you not have to know exact bowden length. You  not need these hundred of speeds definied and prime and so on. Also if your filament may slip by loading you don't have problem because the fixed definied length is not matching anymore... Should be also less code...

### Feature Workflow

Variant a)
Simple filament switch manually:

1. Lift z by amount X and pause print
2. Disable e-stepper
3. May add a security timer for e-heater turn off
4. By user input (click or gcode) drive z back down by amount X and resume print ( if heater was off first heat )

Variant b)
Simple filament switch automatically:

1. Lift z by amount X and pause print
2. May add a security timer for e-heater turn off
3. By user input unload filament till user input ( if heater was off first heat )
4. By user input load filament till user input ( if heater was off first heat )
5. By user input resume print ( if heater was off first heat )

Thank you

rs-development, 2020-02-25:
Sounds very useful for me.

qwewer0, 2021-01-01:
I think this can be closed, as we have `ADVANCED_PAUSE_FEATURE`, and https://github.com/MarlinFirmware/Marlin/pull/20346

> This prevents the extruder from locking up and preventing manual filament advancement during filament change operations.

sjasonsmith, 2021-01-01:
I agree that this seems to be implemented. If there are outstanding issues they should probably have new reports based on the current state of Marlin.

closed 2021-01-01 by sjasonsmith