MarlinFirmware/Marlin #16954: [BUG] Same servo for probe and Nozzle Switch angles mixed up, not used.

danandmary, 2020-02-25:
### Bug Description

My setup uses a single servo for both switching nozzles and lowering the probe (both servo 0).  Set up angles in firmware for both nozzle switching and servo probe, however M502 M500 loads nozzle switch angles only.  When homing G28,  nozzle switch angles are used to deploy and retract probe.  T0 and T1 also uses the same angles.  

### My Configurations
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4247864/configs.zip)

### Steps to Reproduce

1) Setup angles for Probe deploy and retract (85,120)
2) Setup angles for Nozzle Switch E0 and E1 (120,60)
3) Compile firmware, run M502 and M500.  
4) Run M503 "Servo Angles:
12:36:31.995 : echo:  M281 P1 L120 U60"
5) Home printer, Servo switches to E0 (120) position and crams nozzle into bed.  If I click probe manually, after homing z, Switches to E1 position (60).  

**Expected behavior:**
I expect when homing or probing moves are called Servo Probe deploy and retract values are respected.  Likewise when issuing T0 and T1, Nozzle switch angles are respected

danandmary, 2020-02-25:
More Information : 
Defined multiple servos - Switching Nozzle (Servo 1) and Servo Probe (Servo 0).  Mapped Servo 1 Pin to the same as Servo 0.  
Now after loading firmware run M502 then M500
Next Run M281 P0
echo: Servo 0 L120 U60
and M281 P1
echo: Servo 0 L0 U0

As you can see, it seems the servo angles for the #define Z_SERVO_ANGLES { 85, 120 } are not being loaded : 

From the firmware :
#define Z_PROBE_SERVO_NR 1       // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES { 85, 120 } // Z Servo Deploy and Stow angles

danandmary, 2020-02-25:
Further to all of this, when I comment out : #define EDITABLE_SERVO_ANGLES
and run G28, the Servo 0 (Switching Nozzle positions) are used instead of SERVO 1  (Servo Prove) positions.

thinkyhead, 2020-02-25:
What happens if you correct this setting and set it to 2?
```cpp
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
```

thinkyhead, 2020-02-25:
Never mind that. What you see is not a bug, but simply not a supported combination at this time. You can have a servo for a `SWITCHING_EXTRUDER` or a `SWITCHING_NOZZLE` or a Servo probe / `BLTOUCH`, but these cannot be combined because the code was never written to handle it.

thinkyhead, 2020-02-25:
Please give #16960 a test and see if it resolves the issues.

closed 2020-02-26 by thinkyhead

Cultivate3D-Dan, 2020-03-18:
Im sorry it took so long.  So heres what happens : 
Merged Current BF 2.0
M502
M500
G28 > Printer Homes X and Y Fine but when it comes to "deploy" the probe it uses the T0 and T1 angles to stow / deploy.  Pressing probe manually to continue testing
T0 > Switches to T0 Servo position however offset movement does not occur (currently set to X20mm, Y5mm)
T1 > As above switches to T1 Servo position, likewise offset movement does not occur.

Expected behavior  :: 
G28>, X Homes, Y Homes, Servo Moves to probe position, Probe triggers, Servo Moves to Retract probe position. 
T0 > Servo moves to T0 angle, X Y and Z move according to offset
T1 > Servo moves to T1 angle, X Y and Z move according to offset

If I try to define 2 servos ie : NUM_SERVOS 2 (both having the same pin def) :: The servo twitches.  Looking at signal with Oscilloscope, can see signal bouncing around.