MarlinFirmware/Marlin #16958: [BUG] (Babysteping not working both z+and z-)

Bradley-Liu, 2020-02-25:
### Babysteping not working both z+and z-

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4248075/config.zip)

### Steps to Reproduce

1. start printing something
2. double click to begin baby stepping

**Expected behavior: the nozzle should move up and down according to my action on lcd

**Actual behavior: the nozzle doesn't move at all. btw, i have my softend z disabled by changing the menu item. i have enabled the softend on/off menu item in the firmware.

#### Additional Information

thisiskeithb, 2020-02-25:
Have you tried [`2.0.4.3`](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.4.3)  released earlier today? Baby stepping was fixed in that update.

thinkyhead, 2020-02-25:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2020-02-25:
Duplicate of #16941

closed 2020-02-25 by thinkyhead

Bradley-Liu, 2020-02-25:
not yet.  i downloaded the firmware from github four days ago. i am new to marlin. can i still use the config files from the older versions? Do i need to make the changes on the new config files? I make a lot of changes to the configs i don't want to redo my changes if possible.

thinkyhead, 2020-02-25:
We haven't altered the configs significantly in the last 4 days.

Bradley-Liu, 2020-02-25:
I test it and the bug is fixed. but when i define PROBING_FANS_OFF , my hotend fan is still blowing.
i just execute the following code. m28 and m29 is written in test,gcode. i just typed out for your convenience
M23 /test.gcode
M24 
M28
M29

thinkyhead, 2020-02-25:
Are you referring to your `E0_AUTO_FAN_PIN` fan?