MarlinFirmware/Marlin #16961: Old/Forgotten "TRINAMICS" defintions in Conditionals_post.h

nklrk, 2020-02-25:
In earlier releases there was an #define TRINAMICS in Conditionals_post.h. This define was replaced by "HAS_TRINAMIC || HAS_TRINAMIC_STANDALONE". But in line 648 and 668 are two old "TRINAMICS" leftovers. So I guess they also should be replaced with "HAS_TRINAMIC || HAS_TRINAMIC_STANDALONE".
By now this results in setting up wrong values for "MINIMUM_STEPPER_PULSE" and "MAXIMUM_STEPPER_RATE" when using TMC drivers. (see Screenshot)
[image]

Greets,
Niko

nklrk, 2020-02-25:
added config files (I thing they are not relevant in this case)
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4249033/config.zip

ellensp, 2020-02-25:
Created a PR to fix this.

thinkyhead, 2020-02-26:
Good catch, thanks!

closed 2020-02-26 by thinkyhead