MarlinFirmware/Marlin #16964: MKS_SGEN_L freezes at Marlin boot logo when STEP or DIRr pins of unused second are connected to 10k pulldown.

hamster65, 2020-02-25:
I have replaced my Arduino Mega+Ramps with an MKS_SGEN_L board. Since I only have one extruder, I wanted to use the free pins of the unpopulated second extruder for controller and extruder fan. I have N-Channel Mosfets with 10k pulldowns.

When I connect them to  STEP and DIR, Marlin freezes at the boot logo. With only one of the pins connected, sometimes Marlin boots. The chance is higher when I disconnect the USB cable. I also managed to freeze Marlin at the boot logo by just pressing my fingers down on the pins.

It seems Marlin is reading those pins at startup for whatever reason and crashes when it reads a logic 0. I also measured around 2.5 V at one of those pins. Looks like it is floating. Even with nothing connected to it, I might prevent marlin from booting.

I am not sure how to investigate further. For me, the problem is solved, I simply used different pins (TX of E1 uart and enable of E1).

ellensp, 2020-02-26:
E1_STEP and E1_DIR is also used as RXD3, TXD3
It is also the i2c pins SDA1 and SCL1
Did you enable SERIAL 3 or a I2C device anywhere in firmware?

hamster65, 2020-02-26:
No, I don't use I2C or additional uarts apart from the USB connection.
I'll attach the configs as soon as I can and I will also do some more testing to reproduce this issue.

thinkyhead, 2020-02-26:
Sorry if our template is unclear. We don't want to pester you with a lot of questions, so please supply your Configurations for us to use as reference.

hamster65, 2020-02-27:
ellensp, you were right, I2C was indeed active. My config was (by error) based on  /Configurations-release-2.0.4/config/examples/Mks/Sbase/
#define DIGIPOT_I2C was the culprit.

closed 2020-02-27 by hamster65

thinkyhead, 2020-02-28:
Hopefully a sanity check like this will catch this kind of problem in future:
```cpp
#if ANY(DIGIPOT_I2C, DIGIPOT_MCP4018, DAC_STEPPER_CURRENT, EXPERIMENTAL_I2CBUS, I2C_POSITION_ENCODERS, NEOPIXEL_LED, PCA9632, I2C_EEPROM)
  #define USEDI2CDEV_M 1

  #if USEDI2CDEV_M == 0
    // P0_27 [D57] (AUX-1) .......... P0_28 [D58] (AUX-1)
  #elif USEDI2CDEV_M == 1
    // P0_00 [D20] (SCA) ............ P0_01 [D21] (SCL)
  #elif USEDI2CDEV_M == 2
    // P0_10 [D38] (X_ENABLE_PIN) ... P0_11 [D55] (X_DIR_PIN)
  #endif

  #define USED_SDA _CAT(SDA,USEDI2CDEV_M)
  #define USED_SCL _CAT(SCL,USEDI2CDEV_M)

  #if E1_STEP_PIN == USED_SDA || E1_DIR_PIN == USED_SCL
    #error "One or more i2c pins overlaps with E1 pins! Disable i2c peripherals."
  #endif

  #undef USEDI2CDEV_M
  #undef USED_SDA
  #undef USED_SCL
#endif
```