MarlinFirmware/Marlin #16969: Possible Bug with SKR mini V1.1 with Marlin 2.0.4 , board hangs at upload

Papipapito, 2020-02-25:
Hello

I have a SKR mini v 1.1 , i used originally the marlin provided by BIGTREETECH:
https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1 
Is a 2.xxx compiled in my computer.

Now i get the 2.0.4 from github, and configure only the basic parameters:  

Platformio.ini
I set: STM32F103RC_bigtree

Configuration.h

#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define MOTHERBOARD BOARD_BTT_SKR_MINI_V1_1

Same info of configuration.h of BIGTREETECH

I compile without error, upload the firmware.bin  , and the board No longer responds to usb or anything

I upload the firmware.bin compiled with same computer from BIGTREETECH source, with same options and BOARD (old marlin uses other board name) and work again.

The SD are correct and i put other options into configure.h , TMC2208, and others, but the result is same, board no respons with upload marlin 2.0.4

Any Ideas?

Attach the configure.h clean used into 2.0.0.4
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/4251587/Archive.zip)

Regards

Papipapito, 2020-02-26:
Hello

I solved the problem i set  this env:   STM32F103RC_bigtree_USB. and Work , i can connect USB and see temperature

But not change the name of firmware.bin , and every start with SD inserted, flash the firmware 

TX

closed 2020-02-26 by Papipapito

thinkyhead, 2020-02-26:
> But not change the name of firmware.bin , and every start with SD inserted, flash the firmware

Not sure what's up with that. Maybe the board isn't able to write to the SD?

Papipapito, 2020-02-26:
Hi
I have reviewed it and it works, the problem was that the file already existed and the new update did not overwrite the old file called BIGTREETECH.bigtreetech

Regards