MarlinFirmware/Marlin #16970: [FR] Smarter Manual Pause/Resume

klandingham, 2020-02-25:
### Smarter Pause/Resume functionality.

I am proposing a pause/resume feature very much like what comes with the Ender 3 out-of-the-box. During a print on the Ender 3 (from SD card only), one can select **PAUSE** from the menu. This will stop the print and park the print head away from the bed.  The printer power can now be turned off.  Later, when the printer is powered up again, the firmware examines the SD card for a file containing the state of the printer/print when it was paused. If that is found, the LCD menu offers a **RESUME** menu option. If selected, this will reheat the extruder and bed to the previous temps, return the print head to the spot it was paused and resume printing.

This would be a very useful to add to the firmware - especially if you have a long print and you're not comfortable leaving your printer running unattended.

Perusing the Marlin 2.x source, it seems like most, if not all, of the required functionality is already implemented in the "power loss recovery" feature. The difference would be that the existing power loss protection feature constantly (periodically) writes state data to the SD card (which from what I've read shortens the life of the card), while this feature would write state to the SD card _only_ when a user selects the **PAUSE** option.

### Feature Workflow

1. [**PAUSE** from LCD menu during SD card print]
2. [Printer finishes current layer]
3. [Printer retracts a bit (maybe?)]
4. [Print head parks (home X?) and heaters are turned off]
5. [User powers off printer, leaving SD card inserted]
6. [Later, user powers up printer]
7. [Marlin detects print state data on SD card, offers **RESUME** menu option.]
8. [Print extruder and bed are heated to previous temperatures]
9. [Printer primes extruder a bit (maybe?)
10. [Print head returns to position and print continues]

#### Additional Information

* Please check out this feature on the Ender 3 for a demonstration.
* I'm guessing that the current layer is completed before the actual pause occurs.

rs-development, 2020-02-25:
Would love that feature!

InsanityAutomation, 2020-02-26:
I believe this will function as described if you disable Z raise on parking.

thinkyhead, 2020-02-26:
Marlin is close with the `POWER_LOSS_RECOVERY` feature. You can Pause the print and then just shut down the printer. When you reboot the machine Marlin will ask if you want to resume. At the moment Marlin isn't saving power-loss info at the moment following the park, so the Z will probably be incorrect. But adding that logic to `pause_print` (along with information about how retracted the filament) should get that working.

klandingham, 2020-02-26:
> Marlin is close with the `POWER_LOSS_RECOVERY` feature. You can Pause the print and then just shut down the printer. When you reboot the machine Marlin will ask if you want to resume. At the moment [...]

I guess I'm confused...doesn't POWER_LOSS_RECOVERY constantly write state to the SD card? And if so, isn't that considered "not a good thing to do"?  If those two things are true, then I'm suggesting something different.

> I believe this will function as described if you disable Z raise on parking.

Are you saying selecting "pause print" from the menu already writes state data to the card?

Mysel07, 2020-02-28:
No, function POWER_LOSS_RECOVERY write data to the SD card. The problem is that it writes the current position Z + Z raise on parking. The printer after recovery print  "in the air". I mean the algorithm does not subtract value of Z raise on parking.

klandingham, 2020-02-28:
??? That sounds like it actually doesn't work as it is today?

InsanityAutomation, 2020-02-28:
The above issue is correct and present, on pause the raised height is recorded. I opened an issue noting that awhile ago and have not had time to go in and fix it myself. Its on my list but not high on it.

Tupson444, 2022-12-08:
This is a good idea. There could be a separate option beside POWER_LOSS_RECOVERY, that changes the behavior of pause to not raise Z (only move X or Y), and save current position in the print file. After restarting the printer, it would ask to resume print as with power loss.

It already works if using POWER_LOSS_RECOVERY and park when pause is disabled, but it would be nice to have it independently of it (to write to SD only if paused).