MarlinFirmware/Marlin #16972: [BUG] Pin numbering in pins_BTT_SKR_V1_3.h completely wrong

gilliands, 2020-02-25:
### Bug Description

pins_BTT_SKR_V1_3.h EXP<0..1>_<0..9> pin numbering incorrect
Version 2.0.4.3 Release

pins_BTT_SKR_V1_3.h file uses defines in the form EXP<socket>_<pin> to assign processor IO to the physical connector pins

This is a good idea except the person who defined the pin assignments does not appreciate that there is a long standing convention for numbering pins.  Pins are numbered on odd and even rows not sequentially around the connector.

A correct example can be found here:

https://2.bp.blogspot.com/-AeguLHsEKuI/Uv3TyuQT1aI/AAAAAAAABFU/a-S7XCx7g1Y/s1600/2x5+IDC+Header.png

InsanityAutomation, 2020-02-25:
If my memory is correct, this was provided by the manufacturer. Ill have to check to see if there is silk screening that matches the pins file definition or not.

gilliands, 2020-02-25:
I went off the manufacturer's schematic but have noted that the connectors are rotated 180 degrees in the physical socket.

rs-development, 2020-02-25:
I have the board, can check it if you request.

ellensp, 2020-02-26:
There is no numbering visible on the PCB, the provided circuit diagram does follow the convention in the provided image.  Although this probably working. yes it is labeled incorrectly.

gilliands, 2020-02-26:
I had a go at sorting it out....  I moved the beeper pin from the _common.h (it will break other boards) for a test as it made sense to use the same pin position references instead of the direct IO reference. This is more of an example than a suggestion to adopt.

[pins_BTT_SKR_V1_3.zip](https://github.com/MarlinFirmware/Marlin/files/4252813/pins_BTT_SKR_V1_3.zip)

ellensp, 2020-02-26:
the only change I would make is adding keys to ascii art.
eg
```
/* Default Marlin display
 *              _____                                     _____
 *          NC | 1 2 | GND                            5V | 1 2 | GND
 *       RESET | 3 4 | 1.31(SD_DETECT)     (LCD_D7) 1.23 | 3 4 | 1.22 (LCD_D6)
 *  (MOSI)0.18 | 5 6 < 3.25(BTN_EN2)       (LCD_D5) 1.21 | 5 6 < 1.20 (LCD_D4)
 * (SD_SS)0.16 | 7 8 | 3.26(BTN_EN1)       (LCD_RS) 1.19 | 7 8 | 1.18 (LCD_EN)
 *   (SCK)0.15 | 9 10| 0.17(MISO)         (BTN_ENC) 0.28 | 9 10| 1.30 (BEEPER)
 *              ￣￣￣                                     ￣￣￣
 *              EXP2                                      EXP1
 */
```

InsanityAutomation, 2020-02-26:
If the btt documents follow the same pattern is agree we can't change it unless they do. This is done in several places though so they all need to be consistent. @ellensp you want to open a PR with the comment changes to all of them or should I?

closed 2020-02-26 by thinkyhead