MarlinFirmware/Marlin #16982: [BUG] BIGTREETECH SKR 1.3 Onboard SD not working

albydnc, 2020-02-26:
### Bug Description
Hi everyone,
I've flashed my SKR v1.3 with the Marlin Bugfix 2.0.x and configured it with no issues.
Everything is working fine with USB enabling me to print via Octoprint and Simplify3D.
The problem is that the firmware can't mount the Onboard SD; following the forums I've tried to set 
```
#define SDCARD_CONNECTION ONBOARD
#define SDSUPPORT 
```
with no luck.

Thank you for your help!

ellensp, 2020-02-26:
Please provide full configuration files.

ellensp, 2020-02-26:
When you have #define SDCARD_CONNECTION ONBOARD
If the operating system can see the onboard SD card as a USB drive, marlin cannot see the SD card. They are mutually exclusive.
To enable Marlin's SD card  use Use the lcd menus and use the "Attach Media"  After this has been selected a new menu item "Print from Media" will be added to the menu. and standard list file M20 works as expected, the operating system will not be able to see the SD card.
To restore SD access to the operating system select "Release Media" on the LCD. Marlin will no longer have access to the SD card.

This works fine on my skr 1.3 on todays bugfix. .

thinkyhead, 2020-02-26:
I guess it's a "bug" that we need to document these new flags better someplace.

ellensp, 2020-02-26:
I am presuming it's not something else since no config files...

albydnc, 2020-02-26:
[Archivio.zip](https://github.com/MarlinFirmware/Marlin/files/4257430/Archivio.zip)
Here the configuration.
I've tried to use M22 and M21 but it always says `SD card init fail`

NickTom88, 2020-02-26:
So I had this issue and, for S&Gs, tried enabling the following configuration_adv

#if ENABLED(SDSUPPORT)

  // Some RAMPS and other boards don't detect when an SD card is inserted. You can work
  // around this by connecting a push button or single throw switch to the pin defined
  // as SD_DETECT_PIN in your board's pins definitions.
  // This setting should be disabled unless you are using a push button, pulling the pin to ground.
  // Note: This is always disabled for ULTIPANEL (except ELB_FULL_GRAPHIC_CONTROLLER).
  **_define SD_DETECT_INVERTED_**

Once I enabled it along with the other settings you provided, I was able to open the SD card on the controller. Let me know if this works.

ellensp, 2020-02-27:
Provided configuration files has   //#define SDCARD_CONNECTION ONBOARD
This means the LCD SD card is active, not the Onboard SD card.
The SD on the LCD  works as expected with this configuration on my system.

If I enable #define SDCARD_CONNECTION ONBOARD and since this is in a additional  
#if HAS_SDCARD_CONNECTION clause also set 
#define HAS_SDCARD_CONNECTION 1
The Onboard SD works as expected.

NB They have set #define NO_SD_HOST_DRIVE, so you can't release SD back to operating system.

dreemcatcher, 2020-02-29:
Same problem on SKR 1.1 PRO, it read card then need to install firmware but then cant initialize SD card at all. 
Printer Creality CR-10 Drivers - 2208v3

dreemcatcher, 2020-02-29:
https://github.com/dreemcatcher/CR-10-bad-configs-with-no-sd
My configs.

NickTom88, 2020-02-29:
Just printed using SD card. Using about suggested settings. Also included config files this time.

[Configuration_files 2.0.4.4.zip](https://github.com/MarlinFirmware/Marlin/files/4270874/Configuration_files.2.0.4.4.zip)

MikeEitel, 2020-03-01:
I try with btt skr pro 1.1 and TMC2209.
All kind of versions: lcd / onboard with sd in or out-> in 2. menue NO MEDIA 

Putting sd card in lcd slot shows in 1. sceen: Media insered / removed 
= Card detection works flawless

BUT in all variants i always see in 2.sceen:  NO MEDIA

I can connect to printer but in no variant i can have the onboard media in pc  to download the firmware.bin.

MikeEitel, 2020-03-01:
Since i modified in the pin file not to check for spi lcd:
//#if SDCARD_CONNECTION == ONBOARD && !defined(HAS_SPI_LCD)
#if SDCARD_CONNECTION == ONBOARD

I can store eeprom and see and use gcode files off onboard sd card.
But "removing" the media is still not switching the sd to be accessible from the pc.

Keltere, 2020-03-03:
I can confirm what @MikeEitel suggested. Same exact behavior.

thinkyhead, 2020-03-05:
The `defined(HAS_SPI_LCD)` is wrong and should just be `HAS_SPI_LCD`.

thinkyhead, 2020-03-05:
@NickTom88 — I see that `Conditionals_post.h` contains this:
```cpp
#if HAS_LCD_MENU && DISABLED(ELB_FULL_GRAPHIC_CONTROLLER) && !(defined(ARDUINO_GRAND_CENTRAL_M4) && SD_CONNECTION_IS(ONBOARD))
  #undef SD_DETECT_INVERTED
#endif
```
It looks like the SKR boards based on LPC176x might also apply.

Keltere, 2020-03-05:
@thinkyhead thanks i can confirm that now, without LCD in SPI (not connected on exp1 and exp2), the sd it's working for eeprom but when unmounted it won't be hosted. Is that a normal behavior?

thinkyhead, 2020-03-05:
> when unmounted it won't be hosted

If disconnecting / reconnecting the USB doesn't bring it up then it's a mystery. These boards are as all new to me as they are to the rest of you.

jbmorgado, 2020-03-27:
The bugfix didn't work.

My temporary fix was to add the `#define HAS_SDCARD_CONNECTION 1` statement:
```
#define HAS_SDCARD_CONNECTION 1
#if HAS_SDCARD_CONNECTION clause also set
```

SknerusMcK, 2020-04-15:
@ellensp first solution helps, THANKS!