MarlinFirmware/Marlin #16983: [BUG] Can't compile after "Use a STR_ prefix for non-translated strings" commit

thisiskeithb, 2020-02-26:
### Bug Description

"Use a STR_ prefix for non-translated strings" / e78f607ef33044eecde744e77beea87532817c47 causes a slew of errors:

```prolog
Compiling .pio/build/LPC1768/src/src/module/planner_bezier.cpp.o
In file included from Marlin/src/lcd/menu/menu_tmc.cpp:31:
Marlin/src/lcd/menu/menu_tmc.cpp: In function 'void menu_tmc_homing_thrs()':
Marlin/src/lcd/menu/menu_tmc.cpp:153:58: error: 'MSG_X' was not declared in this scope
  153 |   #define TMC_EDIT_STORED_SGT(ST) EDIT_ITEM_P(int4, PSTR(MSG_##ST), &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, []{ stepper##ST.refresh_homing_thrs(); });
      |                                                          ^~~~
Marlin/src/lcd/menu/menu.h:376:24: note: in definition of macro '_MENU_INNER_P'
  376 |   PGM_P const plabel = PLABEL;                                \
      |                        ^~~~~~
Marlin/src/lcd/menu/menu.h:433:54: note: in expansion of macro '_MENU_ITEM_P'
  433 | #define MENU_ITEM_P(TYPE, PLABEL, V...)              _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                      ^~~~Compiling .pio/build/LPC1768/src/src/module/printcounter.cpp.o
~~~~~~~~
Marlin/src/lcd/menu/menu.h:459:55: note: in expansion of macro 'MENU_ITEM_P'
  459 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)               MENU_ITEM_P(TYPE, PLABEL, ##V)
Compiling .pio/build/LPC1768/src/src/module/probe.cpp.o
      |                                                       ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_tmc.cpp:153:35: note: in expansion of macro 'EDIT_ITEM_P'
  153 |   #define TMC_EDIT_STORED_SGT(ST) EDIT_ITEM_P(int4, PSTR(MSG_##ST), &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, []{ stepper##ST.refresh_homing_thrs(); });
      |                                   ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_tmc.cpp:153:53: note: in expansion of macro 'PSTR'
Compiling .pio/build/LPC1768/src/src/module/scara.cpp.o
  153 |   #define TMC_EDIT_STORED_SGT(ST) EDIT_ITEM_P(int4, PSTR(MSG_##ST), &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, []{ stepper##ST.refresh_homing_thrs(); });
      |                                                     ^~~~
Marlin/src/lcd/menu/menu_tmc.cpp:159:7: note: in expansion of macro 'TMC_EDIT_STORED_SGT'
  159 |       TMC_EDIT_STORED_SGT(X);
      |       ^~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_tmc.cpp:153:58: error: 'MSG_Y' was not declared in this scope
  153 |   #define TMC_EDIT_STORED_SGT(ST) EDIT_ITEM_P(int4, PSTR(MSG_##ST), &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, []{ stepper##ST.refresh_homing_thrs(); });
      |                                                          ^~~~
Marlin/src/lcd/menu/menu.h:376:24: note: in definition of macro '_MENU_INNER_P'
  376 |   PGM_P const plabel = PLABEL;                                \
      |                        ^~~~~~
Marlin/src/lcd/menu/menu.h:433:54: note: in expansion of macro '_MENU_ITEM_P'
  433 | #define MENU_ITEM_P(TYPE, PLABEL, V...)              _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                      ^~~~~~~~~~~~
Compiling .pio/build/LPC1768/src/src/module/servo.cpp.o
Marlin/src/lcd/menu/menu.h:459:55: note: in expansion of macro 'MENU_ITEM_P'
  459 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)               MENU_ITEM_P(TYPE, PLABEL, ##V)
      |                                                       ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_tmc.cpp:153:35: note: in expansion of macro 'EDIT_ITEM_P'
  153 |   #define TMC_EDIT_STORED_SGT(ST) EDIT_ITEM_P(int4, PSTR(MSG_##ST), &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, []{ stepper##ST.refresh_homing_thrs(); });
      |                                   ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_tmc.cpp:153:53: note: in expansion of macro 'PSTR'
  153 |   #define TMC_EDIT_STORED_SGT(ST) EDIT_ITEM_P(int4, PSTR(MSG_##ST), &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, []{ stepper##ST.refresh_homing_thrs(); });
      |                                                     ^~~~
Marlin/src/lcd/menu/menu_tmc.cpp:165:7: note: in expansion of macro 'TMC_EDIT_STORED_SGT'
  165 |       TMC_EDIT_STORED_SGT(Y);
      |       ^~~~~~~~~~~~~~~~~~~
*** [.pio/build/LPC1768/src/src/lcd/menu/menu_tmc.cpp.o] Error 1
```

### My Configurations

[My config](https://github.com/thisiskeithb/Marlin/tree/myskr) (but this affects all my printer configs)

### Steps to Reproduce

1. Rebase config to e78f607ef33044eecde744e77beea87532817c47 & observe errors
2. Rebase config to prior commit (6b9a17be16448459cdbf3447598094fae455e39a) & observe no errors.

**Expected behavior:** Successful compile.

**Actual behavior:** Compile fails.

#### Additional Information

Reverting to 6b9a17be16448459cdbf3447598094fae455e39a with the same configs allows successful compilation.

closed 2020-02-26 by thinkyhead

thisiskeithb, 2020-02-26:
@thinkyhead: Did you need more info? I didn’t attach my configs because I linked them instead.

thinkyhead, 2020-02-26:
This was patched already.