MarlinFirmware/Marlin #16984: [FR] Extruder auto fan inverted logic

Berkays, 2020-02-26:
I connected my hotend fan to the neopixel port of the SKR Mini E3 V1.2 board to close the fan when the hotend is below 50C. With the current code it only runs when temperature is below 50C. Is there any easy way to invert the logic for this?

Berkays, 2020-02-26:
Solved it by editing temperature.cpp as attached.

`#if EXTRUDER_AUTO_FAN_INVERTED
      if (temp_hotend[e].celsius <= EXTRUDER_AUTO_FAN_TEMPERATURE)
        SBI(fanState, pgm_read_byte(&fanBit[e]));
        #else
      if (temp_hotend[e].celsius >= EXTRUDER_AUTO_FAN_TEMPERATURE)
        SBI(fanState, pgm_read_byte(&fanBit[e]));
#endif`

closed 2020-02-26 by Berkays

thinkyhead, 2020-02-26:
More likely you want to look at this if the ON/OFF logic of your fan is inverted. That seems odd, though. A "1" state should enable the fan. You can test this by setting the pin directly with `M42`.

```cpp
    #define _UPDATE_AUTO_FAN(P,D,A) do{                  \
      if (PWM_PIN(P##_AUTO_FAN_PIN) && A < 255)          \
        analogWrite(pin_t(P##_AUTO_FAN_PIN), D ? A : 0); \
      else                                               \
        WRITE(P##_AUTO_FAN_PIN, D);                      \
    }while(0)
```

Berkays, 2020-02-26:
@thinkyhead I used a relay to control the fan since the board has no extra outputs for the hotend fan. So basically 24V --> Relay(controlled with AUTO_FAN_PIN) --> LM2596 --> Hotend Fan.