MarlinFirmware/Marlin #16985: SKR v1.3 with AD8495 doesn't boot

atsage, 2020-02-26:
I've been working on this board for hours trying to figure out why the SD card doesn't get mounted to the host and the COM port doesn't respond.  With the Marlin bugfix and smoothieware firmware from Bigtreetech's github it boots right up.  As soon as I made changes to the 2.0.4.3 config and flash the firmware it would stop working.  It would boot up for about 15-30 seconds with the COM port in device manager (which would not respond to prontorface) then the COM port would disappear and it would show up as a failed device with description "device descriptor not recognized".  I must have changed every setting relating to LCD's, EEPROM, serial ports, SD cards, etc.  Luckily I found #14006 which was similar problems to what I was getting and near the bottom mentioned thermal protection kicking in.  I changed TEMP_SENSOR_0 from -4 to 1 and it booted straight up.  It looked like there was a fix implemented but it apparently does not work with this sensor type.  If you need any configs from me let me know but should be pretty easy to reproduce by just setting TEMP_SENSOR_0 to -4 without a sensor plugged in.  Now that I think about it, when I first flashed the firmware I had the sensor plugged in so it might just be broken in general.  I'll try again with everything connected tonight.

thinkyhead, 2020-02-26:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-26 by thinkyhead

atsage, 2020-02-26:
This isn't a "user-support question".  You closed this less than 2 minutes after I posted, did you read it?

atsage, 2020-02-26:
@thinkyhead ????????