MarlinFirmware/Marlin #16986: [BUG] [2.0.1.4] BLTouch probe fails to respond following UBL bed sampling on SKR-PRO 1.1

DrMistry, 2020-02-26:
I have a genuine BLTouch 3.1, configured and mostly working with 2.0.x (I've tried from 2.0.1 to 2.0.1.4) but following a fresh bed sampling for UBL, the probe stops responding, fails to deploy and the bed lifts in to the nozzle when re-homing at the start of the print (this is a custom build CoreXY).  My mainboard is an SKR Pro v1.1 which works perfectly well in every other respect, except that I moved the part-cooling fan to an alternative pin.  The issue was occurring before that change.

I have working EEPROM, the UBL mesh it's self works perfectly well if I reset the printer following the UBL sampling.  I'm using TMC2130s via SPI and use sensorless homing for X and Y which also works perfectly well.  I've tried setting the mode and configuration of the probe to all available options (except for the voltage, I've confirmed I'm getting +5v both at the board and at the connector end of the cable).  

The probe works correctly every time, including dual Z alignment, unless and until I execute the UBL sample run using G29 P0 -> G29 P3 -> G29 S0 -> G29 F10.0 -> G29 A -> M500.  Once that sequence is run, the probe fails to activate, reset or self-test.  The pin is not stuck, bent, or otherwise damaged.  The entire bed sampling run performs perfectly every time.  The probe works perfectly without this issue on an i3 clone using an Anycubic Trigorilla board.

The sensor signal return is connected to the Z-MIN endstop pin (and associated ground).

Obviously this is an edge case, but since I can't find a solution myself I wonder if there's something in my configuration causing gears to grind which hasn't been seen/tested for before.

I attach config.h and config_adv.h

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4257312/Marlin.zip)

yavinmaster, 2020-02-29:
I'm seeing a similar issue with 2.0.3 on the SKR Pro 1.1 board.  I am using BiLinear bed leveling and when running the bed level, the BL Touch stops deploying the probe after several points.

I've attached a scope to the signal pin on the BL touch and normal behaviour appears to be a 1.5ms width signal (3.3V in amplitude) is normally applied (when the probe is retracted), when the probe is deployed the signal width reduces to about 0.6ms, when it appears to be waiting for the probe to be activated it increases to about 1.1ms.  When the probe fails to deploy the signal changes to a flat 3.3V DC level.

I also have EEPROM enabled (via an external I2C EEPROM) and have the zmin enstop pin connected.  I am using TMC2109 with UART (sense pins cut off on X,Y and Z as I'm not using sensorless homing).

The fault doesn't occur straight away, I have 3 repeats per location and the fault appears to occur after the 14 probe location, which happens to be after the first row of points is completed (I have a 14 x 8 grid set up).

I've tried software mode and #define BLTOUCH_DELAY 900 but I think this is more about issues of sensing the trigger as opposed to the deploy itself.

I will try reducing the grid size and see if it occurs after the same number of points or the same "location".

yavinmaster, 2020-03-01:
After rolling back to a version of my settings which worked I have eventually found the issue (in my case) is related to the XY_PROBE_SPEED.  I originally had this set at 3000 and all worked.  I increased it to 10000 and it reliably fails to deploy the BL Touch on the 15 probe point (which is just after the first Y movement).  To be clear, the issue is the output changes from the 1.5ms pulse to a DC level.  I have tried a figure of 6000 and it appears to still work.  I haven't spent the time to find the exact failure point as I have wasted enough time on this already.

I think this might also be related to #16171 .

DrMistry, 2020-03-01:
Well, I've spent days trying to debug this issue.  In desperation, I tried reverting back to a "basic default" configuration which was known to behave more reasonably, and then selectively re-add configuration items.  I find that setting my TMC2130 microsteps to anything ABOVE 64 causes this issue to recur!  I'm going to investigate further, because this seems a bit odd to me.  I suspect there's some memory management issue, as I was also occasionally seeing LCD display corruption, which has also been resolved using less ambitious microstep settings.  I'll post my findings here when I find something meaningful or useful.

boelle, 2020-06-20:
@DrMistry is the isssue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle