MarlinFirmware/Marlin #16987: [1.1.9] MKS-Gen-L screen messing up

FlailAway, 2020-02-26:
Similar to this https://github.com/MarlinFirmware/Marlin/issues/8787 but with MKS-Gen-L board.

### Bug Description
when using REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, the display slowly breaks up. I have tried shielding cables etc, but nothing helps.

### My Configurations
Mint 19.3 + Cinnamon
Arduino 1.8.9 and 1.8.12
Marlin 1.1.9 - current download
MKS-Gen-L board
Full Graphic controller
Configs attached below, sorry if there are any NSFW comments. I think I got them all before zipping. :)
[Configurations.tar.gz](https://github.com/MarlinFirmware/Marlin/files/4257484/Configurations.tar.gz)

### Steps to Reproduce
1: Use Arduino IDE, Marlin 1.1.9
2: Compile and upload

**Expected behavior:** [What you expect to happen]
Screen stay visually solid

**Actual behavior:** [What actually happens]
Screen characters break up randomly and actively. As in, pixels will get a slightly scrambled then a little while later get set back to normal. It's alive. {grin} 

This was working fine on two different printers with the same hardware configuration.
I download the latest 1.1.9 and now screen slowly breaks up on both printers.

ellensp, 2020-02-27:
You need to add some delays when talking to your LCD
Add the following to your Configuration.h
#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(250)

FlailAway, 2020-02-27:
Thanks Ellen, but I went back to an older version of 1.1.9 with a file-date of March-2019 and it is all good again. Is there a better (less granular) version number to identify this early 2019 version? It seems like there are many versions of version 1.1.9 and they are not all the same.

I compared the configuration.h files and the differences were minor and/or not related to my use. But stable the LCD is again.

FlailAway, 2020-02-27:
OK, I copied the configuration.h and configuration_adv.h from the current 1.1.9 into the 2019 Marlin 1.1.19 folder overwriting the old ones and all is still happy with the LCD and I have all of my confugrations as I had set them a few days back when I changed extruder.

Seems something is wrong with 1.1.9 but OK in 1.1.9. :)

thinkyhead, 2020-02-28:
The ghost in the machine.

closed 2020-02-28 by thinkyhead