MarlinFirmware/Marlin #16988: [BUG] Upload for SKR 1.3 suddenly fails in VSCode due to error in /HAL_LPC1768/

JAdams12, 2020-02-26:
I suddenly can't upload firmware, this is 2.0.4.3, the firmware uploaded just fine yesterday, today I changed the PID values for bed and hotend and changed my probe x/y speed because it was too fast and now this is the error I get when trying to compile in VSCode. Everything is updated, it just randomly stopped working!

Thank you for the help!

`Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.2
 - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.6.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode
upload disk:  F:
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\HAL_LPC1768\MarlinSerial.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:755:1: error: expected declaration before '}' token
 }
 ^
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../../Configuration.h:755:1: error: expected declaration before '}' token
 }
 ^
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:755:1: error: expected declaration before '}' token
 }
 ^
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:755:1: error: expected declaration before '}' token
 }
 ^
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o] Error 1
======================================================================================== [FAILED] Took 5.00 seconds ========================================================================================`

psavva, 2020-02-26:
You have a syntax error in your config file:
Marlin\src\HAL\HAL_LPC1768../../inc/../../Configuration.h:755:1: error: expected declaration before '}' token
}

Please attach your config files as instructed when reporting issues

JAdams12, 2020-02-29:
@psavva Thank you so much, not only for solving the issue but for teaching me how to decipher where the error is when reading the error codes. I had no clue that it showed the line there. 

I had the } on the next line after my acceleration settings. I must have bumped enter. 

Thank you again!

closed 2020-02-29 by JAdams12