MarlinFirmware/Marlin #16989: [BUG] LPC1768 h/w pwm servo conundrum

Andrei-Pozolotin, 2020-02-26:
@p3p Chris, @thinkyhead Scott: 
I am hitting a wall with simple issue, any hints are much appreciated

1. **Story**:
after a long while I upgraded from old bugfix-2.0.x to new bugfix-2.0.x,
many things got broken, most of them are fixed now, 
except "LPC1768 h/w pwm servo conundrum"

2. **Problem**: 
`M280 P1 Snnn` is not producing any output

board is using 4 servo:
```
#define SERVO0_PIN         P4_28 // BLTouch control signal "BL/PWM"; s/w pwm
#define SERVO1_PIN         P3_25 // Chamber "SRV1"; h/w pwm channel #2
#define SERVO2_PIN         P3_26 // Chamber "SRV2"; h/w pwm channel #3
#define SERVO3_PIN         P1_23 // Chamber TODO;   h/w pwm channel #4
```
3. **Conundrum**:

Points A, B, C, D below seem to contradict each other.

A. Software PWM is working OK, since SEVRO0 works fine for `M280 P0 Snnn`
```
#define SERVO0_PIN         P4_28 // BLTouch control signal "BL/PWM"; s/w pwm
```

B. Hardware PWM is working OK, since FAN works fine for `M106 Snnn`
```
#define FAN_PIN            P2_04  // h/w pwm channel #5
```

C. M43 T - Toggle Pins controls `SERVO1_PIN  P3_25` channel just fine,
for example, the following produces burst of ~50Hz, ~50% duty square wave
```
M43 T S325 L325 R300 W6
```

D. Nonetheless M280 for `SERVO1_PIN  P3_25`  is not producing any output
```
M280 P1 Snnn   // P3_25 stays at 0 volt, no pulses
```

4. **Custom config**:

[Marlin/src/inc/MarlinConfigPre.h](https://github.com/random-builder/Marlin/blob/83533ddc822bb88b4c4a31004ee07af7562698cf/Marlin/src/inc/MarlinConfigPre.h#L39)

[Marlin/src/inc/MarlinConfig.h](https://github.com/random-builder/Marlin/blob/83533ddc822bb88b4c4a31004ee07af7562698cf/Marlin/src/inc/MarlinConfig.h#L34)

[Marlin/user/Wanhao_D6_MKS_SBASE](https://github.com/random-builder/Marlin/tree/83533ddc822bb88b4c4a31004ee07af7562698cf/Marlin/user/Wanhao_D6_MKS_SBASE)

Thank you.

gilliands, 2020-02-27:
I use these pins for RGB illumination and I find that both P3 pins are not initialised correctly on start and the only way I can get them to function as PWM is by switching the illumination to off and then back on again. Just changing the value makes no difference to the 0 DC output. Then after the off/on sequence everything woks as expected with full PWM capability across all three pins (the two P3 pins and the default servo which in my SKR 1.3 is P0.0).  It's a slightly different issue but it relates to the same P3 pins not doing PWM as they used to some time before December last year.

thinkyhead, 2020-02-28:
> they used to some time before December last year.

If you re-compile the older Marlin using the latest LPC1768 framework does it still work? That would tell us whether or not recent changes in the framework could be throwing a wrench into the works.

thinkyhead, 2020-02-28:
Please enable the `PINS_DEBUGGING` feature and use `M43` to print out a list of all the pins so we can see whether any other pin conflicts might be happening. You can paste the output into a reply with ` ``` ` (three backticks) on lines by themselves before and after the text and it will
```
 be nicely formatted.
```

Andrei-Pozolotin, 2020-02-28:
here is `M43` and `M43 I` report: [report_m43.md](https://github.com/random-builder/Marlin/blob/0cad19074496e9a8ecab886d3081e509551d2dbc/buildroot/user/report_m43.md)

ellensp, 2020-02-28:
```
Recv: PIN: 3.25        SERVO1_PIN                             Input  = 0
Recv: PIN: 3.26        SERVO2_PIN                             Input  = 0
Recv: PIN: 4.28        SERVO0_PIN                             Input  = 0
```
input? on a servo?

Andrei-Pozolotin, 2020-02-28:
@ellensp 
this is how `M43` reports it, and `PIN 3.25` works fine as output, see point `C` for `M43 T` above

thinkyhead, 2020-02-28:
Marlin calls `HAL_init` as the very first thing in `setup()` and that function initializes the servo pins like this:
```cpp
  // Init Servo Pins
  #define INIT_SERVO(N) OUT_WRITE(SERVO##N##_PIN, LOW)
  #if HAS_SERVO_0
    INIT_SERVO(0);
  #endif
  #if HAS_SERVO_1
    INIT_SERVO(1);
  #endif
  #if HAS_SERVO_2
    INIT_SERVO(2);
  #endif
  #if HAS_SERVO_3
    INIT_SERVO(3);
  #endif
```
…right after initializing the LEDs.

So perhaps we need to move servo initialization someplace later in the setup.

Also of note in the LPC1768 version of FastIO these are defined:
```cpp
#define SET_DIR_INPUT(IO)     LPC176x::gpio_set_input(IO)
#define SET_DIR_OUTPUT(IO)    LPC176x::gpio_set_output(IO)

#define SET_MODE(IO, mode)    pinMode(IO, mode)
```
But the SET_MODE macro is never used. Only the `SET_DIR_(IN|OUT)PUT` macros are applied. That might not be important, but you could try this in `HAL_LPC1768/main.cpp`…
```cpp
  // Init Servo Pins
  #define INIT_SERVO(N) do{ pinMode(SERVO##N##_PIN, OUTPUT); OUT_WRITE(SERVO##N##_PIN, LOW); }while(0)
```

Andrei-Pozolotin, 2020-02-28:
@thinkyhead 

1. basic pin init seems not a problem, as confirmed by:
```
M43 T S325 L325 R300 W6 // works fine, produces square wave burst
```

2. nonetheless, I tried with `pinMode`, it made no difference:
```cpp
  // Init Servo Pins
  #define INIT_SERVO(N) do{ pinMode(SERVO##N##_PIN, OUTPUT); OUT_WRITE(SERVO##N##_PIN, LOW); }while(0)
```

3. is there a way to verify that actual **PWM configuration registers** are loaded properly?

thinkyhead, 2020-02-28:
So that implies initializing the servo pins later might help. Try sticking the servo init code into the end of `setup()` to see if it works in that last possible moment.

Andrei-Pozolotin, 2020-02-28:
@thinkyhead 
1. I applied "tail init" as follows:
[Marlin/src/HAL/HAL_LPC1768/main.cpp](https://github.com/random-builder/Marlin/commit/3679c60d3a53b577c32817100f2ed4e0c9866074#diff-71ea14de63044140a1b38e2fe9c73cbd)
2. I still see same 3 symptoms:
a) after reset, SERVO1_PIN 3.25 stays low at 0 volt
b) after `M280 P1 Snnn`  there is no reaction, still at 0 volt
c) after `M43 T S325 L325 R300 W6` there is a square wave burst

Andrei-Pozolotin, 2020-02-28:
I also noticed in [Marlin/src/module/servo.cpp](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/servo.cpp#L39)

1. there is `attach()` then `detach()`
```
  #if NUM_SERVOS >= 1 && HAS_SERVO_0
    servo[0].attach(SERVO0_PIN);
    servo[0].detach(); // Just set up the pin. We don't have a position yet. Don't move to a random position.
  #endif
```

2. but then there is no more `attach()` anywhere

Andrei-Pozolotin, 2020-02-28:
extended "tail init" like this:
https://github.com/random-builder/Marlin/commit/ba7a55dc7294e0b5e4bc42fa7d45c8ca1f030b09
also makes no difference

Andrei-Pozolotin, 2020-02-28:
perhaps we should introduce "init servo" `M` g-code :-)

Andrei-Pozolotin, 2020-03-08:
@thinkyhead Scott: followup: problem found (maybe?)

1. the following M43 hack:
[M43 hack 1](https://github.com/random-builder/Marlin/commit/1fc01d9eae9477ea59e4d4e1bb0b5112dfbd2b30)

2. produces a report:
[report_m43_hack_1.md](https://github.com/random-builder/Marlin/blob/1fc01d9eae9477ea59e4d4e1bb0b5112dfbd2b30/buildroot/user/report_m43_hack_1.md)

which shows that:
* P3_25 is mapped to pwm channel 1
* P3_26 is mapped to pwm channel 2

while per lpc1768 spec
https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf
it is supposed to be instead:
* P3_25 is mapped to pwm channel 2
* P3_26 is mapped to pwm channel 3

3. given that possible mappings are:
```
// Hardware channel  ->  Mappable pins
// 1 ->  P1_18, P2_00
// 2 ->  P1_20, P2_01, P3_25
// 3 ->  P1_21, P2_02, P3_26
// 4 ->  P1_23, P2_03
// 5 ->  P1_24, P2_04
// 6 ->  P1_26, P2_05
```

4. then also implies that default leds pins mapping from 
`Marlin/src/pins/lpc1768/pins_MKS_SBASE.h`
```
#define LED3_PIN           P1_20
#define LED4_PIN           P1_21
```
could possibly interfere with pwm channel mapping?

5. nonetheless, when leds are `#undef`-ed:
[undef leds 2 3 4](https://github.com/random-builder/Marlin/commit/acdb0afc2fc58c7509cf0d5fafe387436fd859ea)
the problem is still not resolved, `M280 P1 Snnn` is inactive

6. finally, channel numbering 1 ... 6
[system/lpc176x/pin_control.h](https://github.com/p3p/pio-framework-arduino-lpc176x/blob/master/system/lpc176x/pin_control.h#L263)
appears to be consistent between the spec and the framework library
which means that the above report does show a problem

7. so the conundrum continues ...

thinkyhead, 2020-03-11:
Are we looking at a platform error? Or is it an error in our fastio? Is there a distinction between 1768 and 1769 we are forgetting?

I'll look at this more closely after some needed sleep….

sci-phi, 2020-04-18:
Is this reason I cant use Servo 0 (P2_00) on SKR 1.3 as PWM? 

[...]
Send: M43 P200
Recv: PIN: 2.00        SERVO0_PIN                             Input  = 1
Recv: ok
[...]
Send: M42 P200 S128
Recv: ok
[...]
Send: M43 P200
Recv: PIN: 2.00        SERVO0_PIN                             Output = 0
Recv: ok

Nothing happened :( I expected to hear fan spin up :(

Caveat: I am not super proficient at this...

I am trying to drive an RRD Fan Extender with 12V continuous power from upper left socket to control a third fan to cool the TMC2208 stepper drivers when in use. Currently 'fan3' is on that continuous power (and full blast 24/7)

I have second fan on HE1 and this is a delta so the Max endstops are unavailable for me to use, as well as Z Min (used for Z probe) - with the diag pins on X&Y Min that leaves just Servo0 as a PWM output ... as I understand it.

boelle, 2020-06-20:
@Andrei-Pozolotin is the issue still there?

Andrei-Pozolotin, 2020-06-21:
resolved by switching from marlin to klipper

closed 2020-06-21 by Andrei-Pozolotin