MarlinFirmware/Marlin #16990: [BUG] 2.0.4.4 Build fails for STM32F103RC with default config

swilkens, 2020-02-26:
### Bug Description

Marlin build 2.0.4.4 build fails for env "STM32F103RC_bigtree" with standard config files.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4258175/Marlin.zip)

### Steps to Reproduce

1. Download Marlin 2.0.4.4
2. Download marlin configurations "release-2.0.4"
3. Change "default_envs" in platformio.ini to "STM32F103RC_bigtree"
4. Build 

**Expected behavior:** [What you expect to happen]
Firmware.bin output

**Actual behavior:** [What actually happens]

[error.txt](https://github.com/MarlinFirmware/Marlin/files/4258189/error.txt)

swilkens, 2020-02-26:
Just to verify, 2.0.4.3 builds just fine.

Scope666, 2020-02-26:
Similar error compiling with LPC1768:

```
error: 'MSG_SD_ERR_READ' was not declared in this scope
Compiling .pio\build\LPC1768\src\src\gcode\sdcard\M26.cpp.o
```

`Marlin\src\gcode\../inc/../core/serial.h:251:59: error: expected ')' before string constant`

GMagician, 2020-02-26:
try bugfix

swilkens, 2020-02-27:
I can confirm current bugfix-2.0.x (https://github.com/MarlinFirmware/Marlin/tree/bfbd976c09128725644fabb96d2153774c22460c) fixes this. Thanks for the heads up @GMagician 

closing

closed 2020-02-27 by swilkens