MarlinFirmware/Marlin #16991: [BUG] planner.cpp error: 'maxj' was not declared in this scope (bugfix 2.0.x)

Kokorone, 2020-02-26:
### Bug Description

Compiling bugfix 2.0.x with todays commits causes this error with a SKR 1.3

[image]

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4258228/Marlin.zip)

### Steps to Reproduce

1. Configure bugfix 2.0.x with commits as of 1:23 pm PST on 2/26/2020
2. Attempt to compile 
3. View errors in terminal

**Expected behavior:** [What you expect to happen]
Compile as usual with no errors
**Actual behavior:** [What actually happens]
Fail with several errors related to maxj, axis, and extraxyjerk in planner.cpp
#### Additional Information
Only noteworthy change is using CLASSIC_JERK
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Kokorone, 2020-02-26:
Recompiled firmware with CLASSIC_JERK disabled and Junction Deviation re-enabled and it compiled with no issues.

Kokorone, 2020-02-26:
Appears to have been broken with this commit which references the exact lines that are erroring

[https://github.com/MarlinFirmware/Marlin/commit/263a7beeb53a619cc319be2e369ef7604f32af4d](url)

lesstutrey, 2020-02-26:
having same issue.

lesstutrey, 2020-02-26:
[image]

mak0t0san, 2020-02-27:
Getting this as well.  The maxj variable was declared as a float variable, but now isn't declared at all.  Same with `axis`.  I'm using a delta printer so I can only use classic jerk as junction deviation doesn't work with delta printers.

ellensp, 2020-03-21:
This was fixed three weeks ago. Confirmed that your configuration files currently compile without issue.
Please close this issue.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle