MarlinFirmware/Marlin #16992: 2.0.4.4

shawnp30, 2020-02-26:
using btt gtr v1.0 with 2.0.4.4 when enabling sdcard support i get errors with sd card string. when i disable eeprom and sd card supoort marlin compiles 
[image]

szepnorbee, 2020-02-26:
+1 with BOARD_BTT_SKR_V1_3

GMagician, 2020-02-26:
Try bugfix

GMagician, 2020-02-26:
#16990 duplicate

shawnp30, 2020-02-26:
tried bug fix still error

freeasabeer, 2020-02-26:
In file queue.cpp line 530, replace MSG_SD_ERR_READ by STR_SD_ERR_READ

lesstutrey, 2020-02-26:
freeasabeer that sort of works.. but just leads to new compile errors in planner.cpp so..

freeasabeer, 2020-02-26:
Not on my side: melzi_optiboot with Creality Ender 3 configuration, the build ends successfully with the fix I proposed.

oscar6502, 2020-02-26:
freeasabeer, works for me, thanks!

wkelley70, 2020-02-26:
> In file queue.cpp line 530, replace MSG_SD_ERR_READ by STR_SD_ERR_READ

Worked for me with a BTT SKR 1.3.  Thanks.

thisiskeithb, 2020-02-26:
This has been fixed in `bugfix-2.0.x`:
https://github.com/MarlinFirmware/Marlin/blob/bafe8d920a780bc5f5c0ad41437a09ffa823432d/Marlin/src/gcode/queue.cpp#L530

Scope666, 2020-02-27:
> This has been fixed in `bugfix-2.0.x`:
> https://github.com/MarlinFirmware/Marlin/blob/bafe8d920a780bc5f5c0ad41437a09ffa823432d/Marlin/src/gcode/queue.cpp#L530

New error after applying this fix:

```
Marlin\src\module\planner.cpp: In static member function 'static bool Planner::_populate_block(block_t*, bool, const abce_long_t&, const xyze_pos_t&, feedRate_t, uint8_t, const float&)':
Marlin\src\module\planner.cpp:2407:19: error: 'maxj' was not declared in this scope
                   maxj = (max_jerk[axis]          // mj : The max jerk setting for this axis
                   ^~~~
Marlin\src\module\planner.cpp:2407:19: note: suggested alternative: 'fmaxl'
                   maxj = (max_jerk[axis]          // mj : The max jerk setting for this axis
                   ^~~~
                   fmaxl
Marlin\src\module\planner.cpp:2407:36: error: 'axis' was not declared in this scope
                   maxj = (max_jerk[axis]          // mj : The max jerk setting for this axis
                                    ^~~~
Marlin\src\module\planner.cpp:2407:36: note: suggested alternative: 'exit'
                   maxj = (max_jerk[axis]          // mj : The max jerk setting for this axis
                                    ^~~~
                                    exit
Marlin\src\module\planner.cpp:2396:23: warning: unused variable 'extra_xyjerk' [-Wunused-variable]
       constexpr float extra_xyjerk = 0;
                       ^~~~~~~~~~~~
Compiling .pio\build\LPC1768\src\src\module\servo.cpp.o
Compiling .pio\build\LPC1768\src\src\module\stepper.cpp.o
*** [.pio\build\LPC1768\src\src\module\planner.cpp.o] Error 1
```

Kokorone, 2020-02-27:
> New error after applying this fix:
>
> ```
> Marlin\src\module\planner.cpp: In static member function 'static bool Planner::_populate_block(block_t*, bool, const abce_long_t&, const xyze_pos_t&, [...]

I have this bug already submitted under this issue link. It only occurs when you have CLASSIC_JERK enabled

[https://github.com/MarlinFirmware/Marlin/issues/16991]url

zonalimitatore, 2020-02-27:
> 
> 
> In file queue.cpp line 530, replace MSG_SD_ERR_READ by STR_SD_ERR_READ

Thanks.Work on atmega gt2560

Danielstc91, 2020-02-27:
> In file queue.cpp line 530, replace MSG_SD_ERR_READ by STR_SD_ERR_READ

Also worked for me with BTT SKR 1.3.

thinkyhead, 2020-02-27:
Thanks for the report. Adding an automated test for EEPROM off with SD on.

closed 2020-02-27 by thinkyhead

wmmc01, 2020-03-02:
This: In file queue.cpp line 530, replace MSG_SD_ERR_READ by STR_SD_ERR_READ, also fixed the error for the BTT SKR V1.4 Turbo!