MarlinFirmware/Marlin #16994: [BUG] ("LEVEL_CORNERS_HEIGHT" should be "RAW_Z_POSITION(0.0)" not "0.0")

thijses, 2020-02-26:
just a simple fix:
currently, "LEVEL_BED_CORNERS" moves to a z height which can be set by "LEVEL_CORNERS_HEIGHT".
the default filled in value is "0.0":
"#define LEVEL_CORNERS_HEIGHT 0.0"
but, this only works if the Z_MIN endstop is at 0.0
The problem is: 
when a home offset is present (like z @ endstop = -1.0),
the "level corners" menu feature will still move the hotend to z_min, which is now -1, 
which obviously destroys the bed and/or your nozzle.
**by changing this to:**
"#define LEVEL_CORNERS_HEIGHT  RAW_Z_POSITION(0.0)"
**the issue is fixed.**

alternatively, you could replace the line:
"line_to_z(LEVEL_CORNERS_HEIGHT);"
in menu_bed_corners.cpp
to:
"line_to_z(RAW_Z_POSITION(LEVEL_CORNERS_HEIGHT));"

thijses, 2020-03-05:
i tested it and it seems like changing the very definition of the line_to_z() command might be best
in the menu.cpp file
it is currently:

void line_to_z(const float &z) {
  current_position.z = z;

and i changed it to:
void line_to_z(const float &z) {
  current_position.z = RAW_Z_POSITION(z);

thinkyhead, 2020-03-06:
The `line_to_z` takes a native Z coordinate as input. All Marlin code works in terms of native (aka "raw) coordinates. And the `LEVEL_CORNERS_HEIGHT` is defined in terms of native machine coordinates.

And Z0 in native machine coordinates is where the nozzle touches the bed.

Your issue is that you've defined a weird Z home offset.
What is the intended purpose of making it -1?

thijses, 2020-03-06:
> The `line_to_z` takes a native Z coordinate as input. All Marlin code works in terms of native (aka "raw) coordinates. And the `LEVEL_CORNERS_HEIGHT` is defined in terms of native machine [...]

i frequently switch nozzles, and changing the physical position of the endstop is more tedious than just leveling one corner and setting a home offset. If the new nozzle is longer than the old one  the height at the Z endstop becomes negative.
Also, i appreciate the response, the clearer title and the explenation, thank you.

thinkyhead, 2020-03-06:
It sounds like you could benefit much from a BLTouch or similar probe. In the meantime, I believe `M206` should do the trick for you.

thinkyhead, 2020-03-07:
It sounds like you could benefit much from a BLTouch or similar probe. In the meantime, you have the correct workaround for your setup:
```cpp
#define LEVEL_CORNERS_HEIGHT RAW_Z_POSITION(0)
```

closed 2020-06-20 by boelle