MarlinFirmware/Marlin #16995: [BUG] G76 - Probe temp compensation error

freaked1234, 2020-02-26:
### Bug Description

Send: G76
Recv: !Probe position unreachable - aborting

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4258746/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4258747/Configuration.txt)

### My Configurations

Both config files attached.
excerpt:

#define PTC_MAX_BED_TEMP 110

// Park position to wait for probe cooldown
#define PTC_PARK_POS_X 50.0F
#define PTC_PARK_POS_Y 50.0F
#define PTC_PARK_POS_Z 100.0F

// Probe position to probe and wait for probe to reach target temperature
#define PTC_PROBE_POS_X 90.0F
#define PTC_PROBE_POS_Y 100.0F

### Steps to Reproduce

Using G76 always produces same error, no matter what "probe pos" and "park pos" is set. Nozzle-Probe offsets are correct. Prints fine, Auto level works fine.

**Expected behavior:** [What you expect to happen]

Temperature compensation calibration via G76 works

**Actual behavior:** [What actually happens]

Send: G76
Recv: !Probe position unreachable - aborting

#### Additional Information

I just dug out my Wanhao Duplicator I3 and did a few upgrades including SKR 1.4, TMC 2130 SPI and an inductive Z-probe.

It auto levels and prints fine but now I wanted to add temperature compensation for the probe by adding a thermistor. I installed a new thermistor to my probe using thermal compound and capton tape and set it up succesfully in Marlin 2.0 Bugfix. It compiled fine and the FW outputs the correct temp values for Probe Temp in addition to Nozzle and Bed temp.

When I run G76 I always get the same error, no matter what settings I apply for "park pos" and "probe pos". 

Homing before doesnt make a difference.

ellensp, 2020-02-27:
Config file is 010109  ie older marlin.  Please try current bugfix 2.0.x

freaked1234, 2020-02-27:
> 
> 
> Config file is 010109 ie older marlin. Please try current bugfix 2.0.x

Oh? How´s that possible, I downloaded and built Marlin Bugfix 2.0.x last week and only edited the configs (not import).

thinkyhead, 2020-02-27:
Hmm, the configs look ok to me — both have 020004 as the version.

ellensp, 2020-02-27:
I'm must be senile..   version is 20004...   not sue how I mucked that up.

thinkyhead, 2020-02-27:
Your `NOZZLE_TO_PROBE_OFFSET` for Y is very large: 60mm.
```cpp
    // Ensure probe position is reachable
    destination.set(
      temp_comp.measure_point_x - probe.offset_xy.x, // 50 - 10 = 40
      temp_comp.measure_point_y - probe.offset_xy.y  // 50 - 60 = -10
    );
    if (!position_is_reachable_by_probe(destination)) {
      SERIAL_ECHOLNPGM("!Probe position unreachable - aborting.");
      return;
    }
```

thinkyhead, 2020-02-27:
Other uses of `position_is_reachable_by_probe` imply that the intention was something like this:
```cpp
    // Point where the probe would be when the nozzle is over the measure point
    destination.set(
      temp_comp.measure_point_x + probe.offset_xy.x, // 50 + 10 = 60
      temp_comp.measure_point_y + probe.offset_xy.y  // 50 + 60 = 110
    );
```
…but I believe the author (and myself) just missed that `position_is_reachable_by_probe` already subtracts the probe offset. So I believe you should change it to this:
```cpp
    destination.set(temp_comp.measure_point_x, temp_comp.measure_point_y);
```
I'll push a patch shortly.

freaked1234, 2020-02-27:
So I just tryed reducing my nozzle offsets - didnt do anything

Then I changed both lines containing the values you posted in G76_m871.cpp and it didnt work either.

    // Disable leveling so it won't mess with us
    #if HAS_LEVELING
      set_bed_leveling_enabled(false);
    #endif

    bool timeout = false;
    while (true) {
      // Move probe to probing point and wait for it to reach target temperature
      destination.set(temp_comp.measure_point_x, temp_comp.measure_point_y);
      do_blocking_move_to(destination);

      SERIAL_ECHOLNPAIR(
        "Bed temp: ", target_bed,
        "; Probe temp: ", target_probe,
        "  Waiting for probe heating."
      );
------------------------------------

    // Ensure probe position is reachable
    destination.set(
      temp_comp.measure_point_x - probe.offset_xy.x,
      temp_comp.measure_point_y - probe.offset_xy.y
    );
    if (!position_is_reachable_by_probe(destination)) {
      SERIAL_ECHOLNPGM("!Probe position unreachable - aborting.");
      return;
    }

----------------------------

      // Move probe to probing point and wait for probe to reach target temp
      destination.set(temp_comp.measure_point_x, temp_comp.measure_point_y);
      do_blocking_move_to(destination.x, destination.y, destination.z);
      SERIAL_ECHOLNPGM("Waiting for probe heating.");
      while (thermalManager.degProbe() < target_probe) {
        idle(
          #if ENABLED(ADVANCED_PAUSE_FEATURE)
            true
          #endif
        );

freaked1234, 2020-02-27:
Oh, I think there is an error in your fix:

    // Ensure probe position is reachable

    if (!position_is_reachable_by_probe(temp_comp.measure_point_x, temp_comp.measure_point_y)) {
      SERIAL_ECHOLNPGM("!Probe position unreachable - aborting.");
      return;
    }

G76 starts now, will report back when i finished everything

thinkyhead, 2020-02-27:
Oh, hi. Have a look at #17005.

freaked1234, 2020-02-27:
Thank you, everything works now, did a full calibration run!

closed 2020-02-27 by freaked1234