MarlinFirmware/Marlin #16999: [BUG] Can't Compile Marlin 2.0.4.4

NovaViper, 2020-02-27:
### Bug Description

I just updated to Marlin 2.0.4.4 however whenever I compile, I get an error like this: https://pastebin.com/eu8zBgB7

### My Configurations

https://github.com/NovaViper/Marlin/tree/ender3/Marlin

### Steps to Reproduce

1. Use the following configurations listed with Marlin 2.0.4.4
2. Try to compile it with PlatformIO

**Expected behavior:** [What you expect to happen]
Not to stop with error and compile firmware correctly

**Actual behavior:** [What actually happens]
The compilation stops and throws out the error shown

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-02-27:
Use bugfix, this error has already been seen and fixed there.

NovaViper, 2020-02-27:
Thanks for the info! I just compiled the bugfix and it's working

closed 2020-02-27 by thinkyhead