MarlinFirmware/Marlin #17001: [BUG] Marlin 2.0.4.3 compiling errors on using TMC

swatkxs, 2020-02-27:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-02-27:
"Errors shown" ?? no errors here

ellensp, 2020-02-27:
Using your Configs and pins file. Compiles without issues on current Marlin Bugfix 2.0.x

swatkxs, 2020-02-27:
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\HardwareSerial.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\HardwareSerial0.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\HardwareSerial1.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\HardwareSerial2.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\HardwareSerial3.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\IPAddress.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\PluggableUSB.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\Print.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\Stream.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\Tone.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\USBCore.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\WMath.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\WString.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\abi.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\main.cpp.o"
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\core.a" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058\\core\\new.cpp.o"
Archiving built core (caching) in: C:\Users\Automan\AppData\Local\Temp\arduino_cache_792345\core\core_arduino_avr_mega_cpu_atmega2560_3ebc7447872ac57b5b094e19c622fd42.a
Linking everything together...
"F:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc" -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560 -o "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_162058/Marlin.ino.elf" "sketch\\Marlin.ino.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\ctrl_access.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\sysclk.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_composite_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_msc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\uotghs_device_due.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\usb_task.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\include\\digipot_mcp4451_I2C_routines.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\include\\i2c_util.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\LCD_pin_routines.c.o" "sketch\\src\\HAL\\HAL_AVR\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fast_pwm.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fastio.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\u8g_com_HAL_AVR_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\EepromEmulation.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\InterruptVectors.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerialUSB.cpp.o" "sketch\\src\\HAL\\HA [... cut ...] ata\\Local\\Temp\\arduino_build_162058" -lm
avr-gcc: error: CreateProcess: No such file or directory

使用 1.19.1  版本的库 U8glib 在文件夹： C:\Users\Automan\Documents\Arduino\libraries\U8glib 
使用 0.6.2  版本的库 TMCStepper 在文件夹： C:\Users\Automan\Documents\Arduino\libraries\TMCStepper 
使用 1.0  版本的库 SPI 在文件夹： F:\arduino-1.8.12\hardware\arduino\avr\libraries\SPI 
使用 1.0  版本的库 SoftwareSerial 在文件夹： F:\arduino-1.8.12\hardware\arduino\avr\libraries\SoftwareSerial 
exit status 1
为开发板 Arduino Mega or Mega 2560 编译时出错。

ellensp, 2020-02-27:
"avr-gcc: error: CreateProcess: No such file or directory"

You have hit the windows command line length limit.  There is no way to adjust this limit.

Only way around this is ti go into Marlin/src/HAL and delete the following directories (make sure you keep HAL_AVR and shared)

HAL_SAMD51
HAL_STM32_F4_F7
HAL_DUE
HAL_LINUX
HAL_STM32
HAL_TEENSY31_32
HAL_ESP32
HAL_LPC1768
HAL_STM32F1
HAL_TEENSY35_36

swatkxs, 2020-02-27:
Thank you for helping me and this seems a problem on Arduino IDE, when I use vscode of platformio, all passed.

ellensp, 2020-02-27:
Yes, platformio will join some directories into libraries then join all the libraries at the end.  Arduino IDE attempts to join all files together at the end in one large very command.

ellensp, 2020-02-27:
If your all sorted, can you close this bug report?

closed 2020-02-27 by swatkxs

thinkyhead, 2020-02-27:
> this seems a problem on Arduino IDE

It is a limitation. We can't get Arduino IDE to ignore certain files depending on the "current target" because it only sees one "target." It would require us to re-package the HALs as libraries — which maybe we could do, but this could make daily development harder.