MarlinFirmware/Marlin #17003: [BUG] SD card not working on Megatronics V3.1 by install Marlin 2.0.4.3

swatkxs, 2020-02-27:
I have compiled both marlin 2.0.4.3 and 1.1.7, confirmed SD card working at 1.1.7, but showing No Media at 2.0.4.3.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4260462/config.zip)

ellensp, 2020-02-27:
Version 2.0.4.4 has been released...  lots of changes. Give this a try.

ellensp, 2020-02-27:
The other question that comes to mind is does the controller notice if you plug a sdcard in or take it out?

swatkxs, 2020-02-27:
still useless on 2.0.4.4, and the controller does not notice when I plug in & out

swatkxs, 2020-02-27:
compiling comes some error on 2.0.4.4
{
	"resource": "/f:/CR-10 Firmware/Megatronics3.1/Marlin-2.0.4.4/Marlin/src/core/serial.h",
	"owner": "cpp",
	"severity": 8,
	"message": "expected ')' before string constant",
	"startLineNumber": 251,
	"startColumn": 59,
	"endLineNumber": 251,
	"endColumn": 59
}

{
	"resource": "/f:/CR-10 Firmware/Megatronics3.1/Marlin-2.0.4.4/Marlin/src/gcode/queue.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'MSG_SD_ERR_READ' was not declared in this scope",
	"startLineNumber": 530,
	"startColumn": 50,
	"endLineNumber": 530,
	"endColumn": 50
}

ellensp, 2020-02-27:
in queue.cpp find  MSG_SD_ERR_READ and change it to STR_SD_ERR_READ
This has been fixed in bugfix....

tpruvot, 2020-02-27:
on 2.0.x branch too (updated a hour ago)

closed 2020-06-20 by boelle