MarlinFirmware/Marlin #17004: BTT SKR 1.4 Turbo print issues (2.0.4.4)

vonryan123, 2020-02-27:
This only started with this board. Printer worked great before. Tried a few versions of marlin 2.0x even 2.04.4 which came out today.
I have tried:
Old and new slice don't matter
Baby steps disabled
Linear Advance disabled (in the firmware as well)
I have set Vrefs anywhere from stock 1.125 to .68
TMC2209's, TMC2100's and DRV8825's All working correctly no jumper issues
I'm am about ready to toss this back to amazon......Printer works but this surface issue is on everything infill, walls, skin/top surface everywhere......
[image]

martend, 2020-02-27:
I did Frankenstein my ET4 with an SKR 1.4 and TFT35 V3.0 it replaced the 1.3 I put in earlier, but I don't have those issues. Using TMC2208 in UART mode running latest Marlin.

vonryan123, 2020-02-27:
I don't know what the issue is anymore I have tried everything.....worked fine before I added the 1.4 turbo so I'm done two days of BS for nothing..

tvixen, 2020-02-27:
I don't see this as a Marlin problem. Why do you ?

vonryan123, 2020-02-27:
Could be the long line of people that have hit or miss issues with TMC2209's and the buggieness of 32bit. I have no idea.

closed 2020-02-27 by vonryan123

thinkyhead, 2020-02-28:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).